fluidd-core/fluidd #836: Add more camera settings

thbl, 2022-08-17:
### Is your feature request related to a problem? Please describe

More control of camera settings like brightness and exposure, could give better image quality.

### Describe the solution you'd like

Add v4l2-ctl settings to fluidd like this octoprint plugin https://github.com/The-EG/OctoPrint-CameraSettings

thbl, 2022-08-17:
would also make it possible to control PTZ web cameras

pedrolamas, 2022-08-17:
Fluidd is just a web interface running on the browser, so doesn't actually have any direct access to perform such actions.

For that to be possible, we would need Moonraker to provide the necessary APIs for the job, so I recommend you first open a Feature Request there for this.

ChaosBlades, 2022-10-11:
Here is the moonraker issue that thbl created. Feel free to give it a thumb up so that moonraker can gauge the desire for this feature.
https://github.com/Arksine/moonraker/issues/488