fluidd-core/fluidd #849: timelapse snapshot image

Kolkman, 2022-08-28:
### Fluidd Version

v1.19.1-3f99573

### Browser

Chrome, Safari

### Device

Desktop PC, Laptop

### Operating System

macOS, iOS

### What happened

On the timelapse 'tab' the preview image in the 'timelapse status' field doesn't load _when_ user authentication is used  (e.g. `force_logins: true`) instead of running fluidd on a trusted_client. 

The cause seems to be an authentication issue.

[image]

### What did you expect to happen

I expect an image with the current snapshot to be shown as an authenticated user. I presume an authorization token needs to be passed along from fluidd.

### How to reproduce

Use `force_logins: true` in the moonraker `[authorization]` section and the preview doesn't show.

### Additional information

Here is a moonraker logs example of the authentication failure:

```
moonraker.log:2022-08-28 09:33:58,810 [web.py:log_exception()] - 401 GET /server/files/timelapse_frames/frame000008.jpg (192.168.178.220): Unauthorized
moonraker.log:2022-08-28 09:33:58,813 [app.py:log_request()] - 401 GET /server/files/timelapse_frames/frame000008.jpg (192.168.178.220) [No User] 12.31ms

```

pedrolamas, 2022-08-28:
Thanks for reporing this @Kolkman, I've managed to reproduce the issue.

We might need to push the fix for this to after 1.20, but possibly on a patch release afterwards like 1.20.1

matmen, 2022-08-28:
This has already been reported [here](https://github.com/mainsail-crew/moonraker-timelapse/issues/56). The reason I didn't add oneshot tokens back then is because it seemed excessive to request a token for every frame, especially when scrubbing through frames.
This also prevents file thumbnails from showing in the file explorer, so oneshot tokens will need to be added there, too.

Ideally the timelapse directory would be whitelisted (so to not require authentication) like the gcode thumbnails are, but I assume this isn't the case because camera feeds are much more private than gcode previews.

closed 2022-10-11 by matmen