fluidd-core/fluidd #863: Layer progress in Printing widget does not update

joshuaboniface, 2022-09-13:
### Fluidd Version

v1.19.1-3f99573

### Browser

Firefox

### Device

Laptop

### Operating System

Linux

### What happened

In the "Printing" widget on the Fluidd main page, there is an indication of the layer status, i.e.

[image]

This progress never updates beyond layer 0 out of however many there are in the print.

Loading the Gcode preview does show the correct layer value, but this does not affect the main print widget, i.e.

[image]

### What did you expect to happen

The layer progress should reflect the actual progress of the print.

### How to reproduce

This occurs on all prints. It might be a misconfiguration, I'm not sure, but it did work at one point earlier in my experimentation with Klipper, but does not now. I'm looking for advice on what to check.

pedrolamas, 2022-09-16:
Thank you for reporting this @joshuaboniface, can I ask that you upload that gcode file so we can take a look?

Also, can I ask what slicer did you use to create the gcode file?

joshuaboniface, 2022-09-19:
Certainly, here it is in gzipped form: [pathway-light_square_abs.gcode.gz](https://github.com/fluidd-core/fluidd/files/9596249/pathway-light_square_abs.gcode.gz) It's created with Cura 5.1. I was suspecting a slicer problem given that it seems to affect everything I've sliced in the last ~month or longer but not sure what - hopefully this helps!

pedrolamas, 2022-09-19:
Thank you @joshuaboniface. According to the file metadata, you used Cura to slice it but set the "output flavour" to "Repetier", when Klipper expects "Marlin" ([source](https://www.klipper3d.org/Slicers.html#set-the-g-code-flavor-to-marlin))

Might not be related, but I would recommend making the change and seeing if that solves the problem.

joshuaboniface, 2022-09-19:
That totally explains it, and a reslice with Marlin set is showing the progress! Probably left over from my experiments with the stock firmware before moving to Klipper. Thanks for the assist.

closed 2022-09-19 by joshuaboniface