fluidd-core/fluidd #877: Add Forced Auto Scroll to resolve issues with windows loosing auto scroll

ChaosBlades, 2022-09-25:
### Is your feature request related to a problem? Please describe

All items that feature auto scroll constantly loose scroll and you have to scroll back down to get it caught back up. Either the scroll simply stops or it looses position and and isn't auto scrolling at the actual end of the log. There is also the situation where you are trying to scroll the entire page and your mouse lands on a window that auto scrolls and you scroll up accidentally.

### Describe the solution you'd like

Add a forced auto scroll where even if you scroll up it will immediatly upon the next update scroll back down to the bottom unless you click a check box to disable auto scroll.

### Describe alternatives you've considered

Fix the current implementation of auto scroll so it does not loose position

### Additional information

Here is a screenshot of TrueNAS SCALE which has the feature I am proposing. You can scroll up and it will stay unless a new update happens in the log at which point it will scroll back down to the bottom unless you check that box.
[image]

ChaosBlades, 2022-10-04:
Just now noticing the down arrow on the console window. Could change that to be a toggle button. When enabled it turns the accent color and the buttons current function is triggered every time the console is updated.
[image]

matmen, 2022-10-05:
I'm not sure I understand what the feature request is here - the console already provides this functionality (automatically scrolls to latest entry unless you manually scroll in it (or disable auto scroll), the arrow button is to jump back to the latest entry). The current auto scroll shouldn't lose position on a new entry.
Are you requesting this functionality to be added to the software update modals?

ChaosBlades, 2022-10-05:
For me it does loose scroll. It is possible that it could be an issue with Brave Browser but I doubt it. It should be just normal Chrome for this function. Also there is the issue with scrolling the entire page and your mouse pointer coming in contact with the console and scroll is lost. Possible this is why I think it looses scroll when I don't realize I am doing this.

This would also simplify things. Instead of two GUI elements just to control auto scroll it is only one.

Yes, that as well. Don't know how many times I am wondering why an update is taking so long just to realize it lost scroll.

matmen, 2022-10-05:
It seems to be working as intended for me (Chrome 106). I don't think it's too much of a hassle to click the scroll-down-button (which re-enables auto scroll) in case of an accidental scroll.

ChaosBlades, 2022-10-11:
> I've experienced some issues here... seems there is some sort of race-condition where this stops working, especially when multiple lines are added... but I haven't had the time to revisit this as of yet.

This appears to be the case. Did some testing. Didn't touch the mouse or anything to interact with the computer, just watched the console.

- Half way through Input Shaper tuning. While running freq sweep It lost scroll.
- Outputting results from Input Shaper is lost scroll again.
- As soon as I press the Home All button it looses scroll.
- As soon as I press the QGL button I see the scroll down arrow flash 3 times but it does not loose scroll.
- As soon as the any QGL pass is done and it does its first correction it looses scroll. So if it takes 2 passes, it looses scroll twice.
- It looses scroll right at the end of running bed mesh.

closed 2022-10-24 by matmen

lord-carlos, 2023-10-05:
My console still constantly loses the auto scroll. On latest chrome with fluidd stable.
I'm not sure what it is. But it's often.

PR #921 (2022-10-24): fix(console): console no longer loses scroll attachment on multiple M118
Only recheck attachment state if `scrollToLatest` is forced, avoids weird bugs where the console trace detaches on multi-line M118s.

Resolves #877, resolves #904, is more carefully thought through than my proposal in #904.