fluidd-core/fluidd #883: Endpoints port removed from URLs in config.json

liborburda, 2022-09-30:
### Fluidd Version

v1.20.1-2bb2209

### Browser

Firefox

### Device

Desktop PC

### Operating System

Windows

### What happened

First of, thank you all for the excellent work! I really love Fluidd.

I have multiple (two) Moonraker instances running on the same RPi. I would like to specify them in `endpoints` array in config.json, so they are pre-configured across all devices.

### What did you expect to happen

Both instances are pre-configured in Fluidd.

### How to reproduce

- Add two Moonraker URLs to `endpoints` array in config.json (same URL, different ports):
```
... Omitted...
  "endpoints": [
    "http://my.domain.local:7125",
    "http://my.domain.local:7126"
  ],
... Omitted...
```
- Reload Fluidd
- Only one instance is visible in Fluidd

### Additional information

I believe this bug is caused by `sanitizeEndpoint` function, which does not include the port in `return`: https://github.com/fluidd-core/fluidd/blob/develop/src/util/sanitize-endpoint.ts#L10 . When port is missing/removed, both URLs look the same. That's why I can see only one printer in Fluidd.

pedrolamas, 2022-10-02:
The `sanitizeEndpoint` is working correctly, as `url.host` includes the port number (`hostname` however does not, but that is not the case here).

We run a check on each of the endpoints to ensure they are working correctly and if not, we hide it from the list.

I would recommend you manually open each of the endpoint addresses in the browser and confirm that they are responding as expected.

matmen, 2022-10-04:
The `endpoints` key in the config file is only used as a list of *possible* backends, and the first one reachable is chosen as the entry point. This just handles automatic printer selection on first load of fluidd (e.g. when fluidd is hosted on a central server instead of on a printer's RPi), it doesn't (currently) add _all_ reachable printers.

napter, 2022-10-21:
@matmen Oh why not, I thought this was the solution to https://github.com/fluidd-core/fluidd/issues/464

pedrolamas, 2022-10-24:
> @matmen Oh why not, I thought this was the solution to #464

That actually, might not be a bad idea at all... 🤔

pedrolamas, 2022-10-24:
This is working as intended, but we might look into making this more generic so users can add multiple printers via this interface on a later release!

closed 2022-10-24 by pedrolamas