fluidd-core/fluidd #889: Unknown command:"Z_OFFSET_APPLY_ENDSTOP"

ksdfpr, 2022-10-03:
### Fluidd Version

1.20.1

### Browser

Firefox, Safari, Edge

### Device

Desktop PC, Mobile device (Phone, Tablet or similar)

### Operating System

Windows, macOS, iOS

### What happened

I get the following error [Unknown command:“Z_OFFSET_APPLY_ENDSTOP” ] when Im trying to update z offset from fluidd interface save button.
I have a delta kossel printer with manual probe system (microswitch), delta printers don’t have a z endstop

### What did you expect to happen

To update z_offset permanently after SAVE_CONFIG

### How to reproduce

Changing values via buttons 
Press save button

[image]

[image]

### Additional information

Maybe for delta printers  the command  Z_OFFSET_APPLY_PROBE works?...

pedrolamas, 2022-10-03:
Thank you for your report @ksdfpr, can I ask that you upload your printer.cfg file please?

ksdfpr, 2022-10-03:
Yes, sure. Please rename .txt to .cfg
[printer.txt](https://github.com/fluidd-core/fluidd/files/9700258/printer.txt)

PR #890 (2022-10-04): fix: check stepper_z exists before applying z_offset
If there is no `stepper_z` in the config, we can't call `Z_OFFSET_APPLY_ENDSTOP`, so just assume we are using a probe and call `Z_OFFSET_APPLY_PROBE` instead.

Fixes #889
closed 2022-10-04 by pedrolamas