fluidd-core/fluidd #897: Allow the exclude/cancel object feature to be clickable on mobile

simpat1zq, 2022-10-07:
### Is your feature request related to a problem? Please describe

Just want to start by saying thank you for Fluidd and for the exclude object functionality. It was the main thing I was missing from the OctoPrint days. Just in the past month, you've saved a few hundred grams of filament from the landfill for me alone. 

The problem I'm having is that I almost always access my printers from my phone. I can view the names of the objects and cancel from there, but when I'm printing 6 of the same thing, and one of them fails, there's really no way to know which one I should cancel based on the name, since they all have the same name. I can easily see on the gcode preview which one I need to cancel, but I can't tap on that.

I might be missing something, so please let me know if I am, but I just couldn't figure out how to do that from mobile. 

### Describe the solution you'd like

Allow tapping on the the gcode viewer cancel object when using a mobile browser. 

### Describe alternatives you've considered

I've tried to cancel based on filename, but I don't see any way to relate object name to the spot on the bed that failed.

pedrolamas, 2022-10-07:
Thank you for the report @simpat1zq, can I ask what phone, OS (and version) and browser are you using?

simpat1zq, 2022-10-07:
Samsung Galaxy A71. 
Android 12. 
Chrome. 

I also tried it on chrome on a galaxy s9. Same result.

PR #898 (2022-10-07): fix: Exclude Object not reacting to mobile inputs
Closes #897
closed 2022-10-07 by matmen