fluidd-core/fluidd #902: Duplicate Console Output

claudermilk, 2022-10-09:
### Fluidd Version

1.20.1

### Browser

Chrome

### Device

Laptop

### Operating System

Windows

### What happened

Console output is getting duplicated. I'm seeing messages displayed twice.

### What did you expect to happen

Just a single line for each message.

### How to reproduce

Simply do anything to get console output.

### Additional information

I checked against a concurrently running Mainsail page on the same printer. I'm not seeing this behavior in Mainsail.

pedrolamas, 2022-10-09:
Thank you for the report @claudermilk, can you please provide a screenshot of the problem? Thanks in advance!

claudermilk, 2022-10-09:
I'm not sure what was going on, but it seems to have stopped. I'll close this out.

closed 2022-10-09 by claudermilk