fluidd-core/fluidd #910: Pressue Advance Calibration Wizard

iwm911, 2022-10-13:
### Is your feature request related to a problem? Please describe

Klipper tunning like pressure advance is done using external tools and gcode files.

### Describe the solution you'd like

it would be nice if we were able to generate a pressure advance calibration pattern from the Fluidd UI like in 
https://realdeuce.github.io/Voron/PA/pressure_advance.html

It could save time and help many new people to easily tune their printer to perform best.

ChaosBlades, 2022-10-31:
This is the way Marlin does calibration. Seems a lot more complicated than what the Klipper docs say. Not sure if this way is better, easier, or just different. Doing it the Marlin way would be easier inside Fluidd because a lot of those parameters could just be pulled from printer.cfg.
https://www.klipper3d.org/Pressure_Advance.html

pedrolamas, 2025-04-06:
Pressure Advance Calibration can be a complicated process as it depends on a lot of variables.

Nowadays, modern slicers have far better support to generate these calibration models than it would ever be possible to do inside Fluidd, so I don't see this being implemented as a Fluidd feature and instead leave it for the slicers.

closed 2025-04-06 by pedrolamas