fluidd-core/fluidd #922: GUI resets to defaults

legend069, 2022-10-23:
### Fluidd Version

v1.20.1

### Browser

Firefox

### Device

Laptop

### Operating System

Windows

### What happened

GUI keeps resting to defaults.

happens when using the GUI - shutdown
after turning the printer back on with the main powerswitch klipper/fluidd loads up and the GUI has been reset to defaults.
this is very frustrating.
i have to arrange the GUI layout, console filters, macros.

### What did you expect to happen

GUI load to what i have set and not be set to defaults.

### How to reproduce

use the GUI -shutdown
turn the main poweroff before the pi and fully shutdown?
not sure how it happens.

pedrolamas, 2022-10-23:
Thank you for your report @legend069.

I should point out the Fluidd is just a web app running in the browser, so all data is stored back on the device running Moonraker (a RaspberryPi is the most common, but that may vary with your own setup)

I haven't seen anyone reporting such behavior before, but what you describe sounds like the file system has been made read-only (an option present on the Raspberry Pi), where all data is stored in a memory drive and discarded on reboot.

legend069, 2022-10-23:
> Thank you for your report @legend069.
>
> I should point out the Fluidd is just a web app running in the browser, so all data is stored back on the device running Moonraker (a RaspberryPi is the most [...]

It doesn't always happen. It has happened from doing the firmware restart thing too.

Could it be related to the Moonraker new file structure?

pedrolamas, 2022-10-24:
> Could it be related to the Moonraker new file structure?

If it wasn't happening before the changes, then it's a strong possibility!

In any case I don't think this is a direct Fluidd issue as we have no direct control of what happens in the file system.

legend069, 2022-10-24:
> In any case I don't think this is a direct Fluidd issue as we have no direct control of what happens in the file system.

oh ok, so it's moonraker the saves the gui layout, console filters and ect to a file?

pedrolamas, 2022-10-24:
Correct, all those changes in Fluidd are stored in Moonraker's database!

legend069, 2022-10-25:
thanks for the help. I'll close issue.

closed 2022-10-25 by legend069