fluidd-core/fluidd #926: follow progress on gcode preview be displayed on header bar

dez1729, 2022-10-26:
### Is your feature request related to a problem? Please describe

When selecting a layer or progress, it cancels the "follow progress", but in order to resume you have to click through to the gears icon menu in order to select it. This should be something that can be selectable on the header bar without going for the extra click.

### Describe the solution you'd like

The "follow progress" should be selectable on the gcode preview header bar.

pedrolamas, 2022-11-04:
Thank you for your FR @dez1729, I agree and quite honestly, I think that whole panel needs an upgrade!

We will look into this the first opportunity we get!

pedrolamas, 2022-11-29:
I've now started some work on this, here are two options I am currently experimenting with:

## Option 1

[image]

## Option 2

[image]

---

Both options will show tooltips on mouse hover, the 1st option seems to be cleaner but the 2nd option is a bit more accessible on the phone (as the buttons are kept closer)

pedrolamas, 2022-11-29:
@matmen @BastelKlug any comments on the above? 🙂

closed 2022-12-02 by pedrolamas