fluidd-core/fluidd #928: camera configuration not saved, and default camera keeps disabling

aschor, 2022-10-27:
### Fluidd Version

v1.16.2-0fa3c9b

### Browser

Firefox

### Device

Desktop PC

### Operating System

Windows

### What happened

before (what?) I had no issue. But now, every time I start my printer, I've got no camera.

When I go to configuration, the default camera is disabled, and I have to enable it ... every time.

I cannot find where fluidd stores it's configurations, maybe there's a readonly file somewhere ? or some other FS issue ? But what file should I inspect ?

### What did you expect to happen

Fluidd's own configuration to survive a reboot !

### How to reproduce

well ... reboot.

pedrolamas, 2022-10-28:
Thank you for reporting this @aschor.

I've ran some tests yesterday and couldn't reproduce this issue, so I ask that you perform the following steps to test:

- Open Fluidd on the browser as you normally do
- Press `F12` to bring up the Developers Tools on the browser
- On the Developer Tools panel, select the "Console" tab
- Now in Fluidd, go to Settings, then Cameras and try to enable the camera again
- Observe if you get any error message on the Dev Tools Console; if so, please update this issue with the messages
- Press `F5` to refresh Fluidd
- Again, observe if you get any error message on the Dev Tools Console; if so, please update this issue with the messages

aschor, 2022-10-29:
Hi, I cleaned my browser's cache, and before that I added a named camera. when first coming on the LOCAL web interface, I got no error but fluidd asked me to add a printer (well, I was already on the printer's web server ...).

Once added and logged in, no error and my named camera was still here, no console error. F5 => no error. 

could you please tell me where the web interface configuration is stored in the filesystem please ? (on the linux side I assume)

aschor, 2022-10-30:
interesting fact : here is the display from mobile app : 
[image]

seems like there are multiple Default camera, whereas the configuration only olds (on my desktop) "Default" and "pi" camera configurations ...

pedrolamas, 2022-10-30:
> could you please tell me where the web interface configuration is stored in the filesystem please ? (on the linux side I assume)

Fluidd doesn't have a database specifically as it is just a web interface. All we do is requests to Moonraker (the server side API) and it indeed has a database; the location of that database will be in accordance to your own setup, but in the latest versions of Moonraker, it is under `~/printer_data/database`

pedrolamas, 2022-11-04:
> interesting fact : here is the display from mobile app : [image]

To be honest, I'm not sure what is going on here...

But I just noticed you are using a very old version of Fluidd, so my recommendation is that you update to the latest version.

Hopefully that will fix the problem, but if now, please access these 2 URLs on your browser and then add here their response:

http://localhost:7125/server/database/item?namespace=fluidd

http://localhost:7125/server/database/item?namespace=webcams

(replace "localhost:7125" with whatever you use to access Fluidd)

aschor, 2022-11-04:
thanks @pedrolamas . I'll consider upgrade as soon as I can find time to backup everything and not need the printer just in case (because you know, upgrading klipper /moonraker ... well ... that starts with reading all deprecations and things removed/changed and they are numerous, so I'm on the "it works don't update" side ^^)

I disabled moonrakers's authorization to not get a 401 response ;

First command : 
```{"result": {"namespace": "fluidd", "key": null, "value": {"cameras": {"activeCamera": "all", "cameras": [{"id": "7ff85844-5e8a-4f53-b7bc-be49fa379bb0", "enabled": true, "name": "Default", "type": "mjpgadaptive", "fpstarget": 15, "url": "/webcam/?action=stream", "flipX": false, "flipY": false}]}, "charts": {"selectedLegends": {"raspberry_pi": true, "heater_bed": true, "heater_bedTarget": true, "heater_bedPower": false, "extruder": true, "extruderTarget": true, "extruderPower": false}}, "console": {"commandHistory": ["SET_KINEMATIC_POSITION x=0 y=0 z=0", "wait_minutes MINUTES=2", "wait_minutes MINUTES=2", "SET_KINEMATIC_POSITION Z=0.3", "M118 coucou", "G4 P660000", "M118 {printer.configfile.settings.stepper_y.position_min}", "m118 toto", "G4 P60000", "m117 toto", "G4 P60000", "SET_KINEMATIC_POSITION X=0 Y=0 Z=0", "BED_MESH_CALIBRATE probe_count=3,3", "BED_MESH_CALIBRATE probe_count=3,3", "BED_MESH_CALIBRATE probe_count=3,3", "BED_MESH_CALIBRATE probe_count=3,3", "BED_MESH_CALIBRATE probe_count=3,3", "BED_MESH_CALIBRATE probe_count=3,3", "GET_POSITION ", "GET_POSITION "]}, "layout": {"layouts": {"dashboard": {"container1": [{"id": "jobs-card", "enabled": true, "collapsed": false}, {"id": "printer-status-card", "enabled": true, "collapsed": true}, {"id": "toolhead-card", "enabled": true, "collapsed": false}, {"id": "temperature-card", "enabled": true, "collapsed": false}], "container2": [{"id": "camera-card", "enabled": true, "collapsed": false}, {"id": "macros-card", "enabled": true, "collapsed": false}, {"id": "console-card", "enabled": true, "collapsed": false}, {"id": "outputs-card", "enabled": true, "collapsed": false}, {"id": "printer-limits-card", "enabled": true, "collapsed": true}, {"id": "retract-card", "enabled": true, "collapsed": false}, {"id": "gcode-preview-card", "enabled": false, "collapsed": false}, {"id": "bed-mesh-card", "enabled": false, "collapsed": false}]}, "diagnostics": {"container1": [{"id": "9e48a932-2a07-42d5-bd45-f854b53ee8e7", "enabled": true, "title": "Speeds", "collapsed": false, "height": 300, "icon": "motion", "axes": [{"enabled": true, "unit": "mm/s", "showLegend": true, "metrics": [{"collector": "printer.motion_report.live_velocity", "name": "Velocity", "style": {"lineStyle": "solid", "lineColor": "#2196f3", "fillOpacity": 0, "displayLegend": true}}, {"collector": "printer.toolhead.max_velocity", "name": "Max Velocity", "style": {"lineStyle": "dotted", "lineColor": "#0075d2", "fillOpacity": 0, "displayLegend": false}}]}, {"enabled": true, "unit": "mm\u00b3/s", "showLegend": true, "max": 20, "metrics": [{"collector": "printer.motion_report.live_extruder_velocity * Math.PI * (printer.configfile.settings.extruder.filament_diameter / 2) ** 2", "name": "Flow", "style": {"lineStyle": "solid", "lineColor": "#b12f36", "fillOpacity": 5, "displayLegend": true}}, {"collector": "12", "name": "Max Flow", "style": {"lineStyle": "dashed", "lineColor": "#820007", "fillOpacity": 0, "displayLegend": false}}]}]}]}, "dashboard-xs-arnaud": {"container1": [{"id": "jobs-card", "enabled": true, "collapsed": true}, {"id": "printer-status-card", "enabled": true, "collapsed": true}, {"id": "toolhead-card", "enabled": true, "collapsed": true}, {"id": "temperature-card", "enabled": true, "collapsed": true}], "container2": [{"id": "camera-card", "enabled": true, "collapsed": false}, {"id": "macros-card", "enabled": true, "collapsed": false}, {"id": "console-card", "enabled": true, "collapsed": false}, {"id": "outputs-card", "enabled": true, "collapsed": false}, {"id": "printer-limits-card", "enabled": true, "collapsed": true}, {"id": "retract-card", "enabled": true, "collapsed": false}, {"id": "gcode-preview-card", "enabled": false, "collapsed": false}, {"id": "bed-mesh-card", "enabled": false, "collapsed": false}]}}}, "macros": {"stored": [{"name": "M600", "visible": false}, {"name": "M900", "visible": false}, {"name": "moonraker_relaisHotendOn", "visible": false}, {"name": "moonraker_relaisLedsOn", "visible": false}, {"name": "moonraker_relaisBedOff", "visible": false}, {"name": "moonraker_relaisBedOn", "visible": false}, {"name": "moonraker_relaisLedsOff", "visible": false}, {"name": "moonraker_relaisHotendOff", "visible": false}, {"name": "check_extruder", "visible": false}, {"name": "moonraker_relaisledson", "visible": false, "disabledWhilePrinting": false, "color": "", "categoryId": "0"}, {"name": "moonraker_relaisledsoff", "visible": false, "disabledWhilePrinting": false, "color": "", "categoryId": "0"}, {"name": "moonraker_relaishotendon", "visible": false, "disabledWhilePrinting": false, "color": "", "categoryId": "0"}, {"name": "moonraker_relaishotendoff", "visible": false, "disabledWhilePrinting": false, "color": "", "categoryId": "0"}, {"name": "moonraker_relaisbedon", "visible": false, "disabledWhilePrinting": false, "color": "", "categoryId": "0"}, {"name": "moonraker_relaisbedoff", "visible": false, "disabledWhilePrinting": false, "color": "", "categoryId": "0"}, {"name": "shutdown_machine", "visible": false, "disabledWhilePrinting": false, "color": "", "categoryId": "0" [... cut ...] 4140-c37e-4671-a576-9ca49aaddb60": {"name": "picam", "service": "mjpegstreamer-adaptive", "targetFps": 15, "urlStream": "/webcam?action=stream", "urlSnapshot": "/webcam?action=snapshot", "flipX": false, "flipY": false, "rotation": 0, "enabled": true}}}}```

=> I guess that means that moonraker's database got corrupt ? so ... is it fixable ? (editable ?)
moonraker database is at ~/.moonraker_database, but "data.mdb" is not a standard text file nor an sqlite file, so I don't know yet how to fix it

pedrolamas, 2022-11-04:
Here's the thing: before Fluidd 1.20 we were using the "cameras" node under the "fluidd" namespace (what you see in the first JSON blob you posted)

Mainsail was using the "webcams" namespace, and when Moonraker recently decided to take those configurations and make it more consistent, we decided to migrate our "fluidd/camera" data to "webcams" - that was done on #839 and released first time on 1.20

So what I see in your screenshot is most definitely the "webcams" data, which doesn't make much sense as you said you are using Fluidd 1.16!

But yes, your "webcams" namespace shows those cameras, and that is the reason for the whole problem here!

If they don't all show in Fluidd, I think you should be able to delete those "Default" cameras one by one (delete, refresh page, repeat) and that should fix the problem!

aschor, 2022-11-05:
> So what I see in your screenshot is most definitely the "webcams" data, which doesn't make much sense as you said you are using Fluidd 1.16!
> 

but it would make sens if fluidd.xyz (wich is up to date) had messed all up ? because I tried to use the online "app", because I have 2 printers ...

pedrolamas, 2022-11-05:
I guess now I get why you have multiple "Default" entries on the "webcams" namespace: from Fluidd 1.20, we read the legacy "fluidd/camera/cameras", migrate that to "webcams" namespace, and then remove the "cameras" (leaving only the "activeCamera"), but somehow this seems to have failed in your system!

I'm not entirely sure how to explain what has happened here, but I guess I need to run some tests and see if I can repro the issue...

> ...I tried to use the online "app", because I have 2 printers ...

Be aware that you can use a local Fluidd system to control multiple printers! Fluidd just needs to be able to bypass Moonraker CORS, nothing more!

pedrolamas, 2023-09-20:
Closing this now as there has been no more contact regarding this matter and we have made considerable changes in the webcams.

If you still need help with this, I would recommend that you try [Klipper Discord server](https://discord.klipper3d.org/) instead.

closed 2023-09-20 by pedrolamas