fluidd-core/fluidd #931: Firefox takes multiple attempts to connect to websockets

nathanielhudson, 2022-10-30:
### Fluidd Version

v1.20.1, also occured on earlier versions.

### Browser

Firefox

### Device

Laptop

### Operating System

Windows

### What happened

When loading Fluidd in Firefox, it stays on the " http://voron2.local Connecting to moonraker..." screen for quite a while (and sometimes gets stuck here). Opening the browser console shows that it it throwing an error:

`Firefox can’t establish a connection to the server at ws://voron2.local/websocket?token=JVQXEOI35IFRUB5JITJMJQLQ4FKKEKVQ.`

`[WEBSOCKET] Connection error: 
error { target: WebSocket, isTrusted: true, srcElement: WebSocket, eventPhase: 0, bubbles: false, cancelable: false, returnValue: true, defaultPrevented: false, composed: false, timeStamp: 5671, … }
​
bubbles: false
​
cancelBubble: false
​
cancelable: false
​
composed: false
​
currentTarget: null
​
defaultPrevented: false
​
eventPhase: 0
​
explicitOriginalTarget: WebSocket { url: "ws://voron2.local/websocket?token=JVQXEOI35IFRUB5JITJMJQLQ4FKKEKVQ", readyState: 3, bufferedAmount: 0, … }
​
isTrusted: true
​
originalTarget: WebSocket { url: "ws://voron2.local/websocket?token=JVQXEOI35IFRUB5JITJMJQLQ4FKKEKVQ", readyState: 3, bufferedAmount: 0, … }
​
returnValue: true
​
srcElement: WebSocket { url: "ws://voron2.local/websocket?token=JVQXEOI35IFRUB5JITJMJQLQ4FKKEKVQ", readyState: 3, bufferedAmount: 0, … }
​
target: WebSocket { url: "ws://voron2.local/websocket?token=JVQXEOI35IFRUB5JITJMJQLQ4FKKEKVQ", readyState: 3, bufferedAmount: 0, … }
​
timeStamp: 5671
​
type: "error"
​
<get isTrusted()>: function isTrusted()
​
<prototype>: EventPrototype { composedPath: composedPath(), stopPropagation: stopPropagation(), stopImmediatePropagation: stopImmediatePropagation(), … }
[consola.browser.js:1:8811](webpack://fluidd/node_modules/consola/dist/consola.browser.js)
    value consola.browser.js:1
    value consola.browser.js:1
    a consola.browser.js:1
    value consola.browser.js:1
    value consola.browser.js:1
    onerror socketClient.ts:103`

This error repeats (with different tokens) every couple seconds up to five times. 

The weird part is that it isn't consistent. Sometimes it connects immediately, sometimes after one connection error, sometimes after more, and sometimes it completely fails to connect. 

Chrome does not show this this issue. Firefox on OS X does not show this issue either.

### What did you expect to happen

Fluidd to connect immediately.

### How to reproduce

Open Fluidd in Firefox on Windows.

### Additional information

Example of a successful connection after three unsuccessful attempts.
[image]

pedrolamas, 2022-11-04:
Thank you for your report @nathanielhudson.

My main development machine runs Windows and I regularly test with Firefox, however I have not been able to reproduce this issue.

The fact that you have no issue with other browsers leads me to believe there is some other issue outside of the control of Fluidd, but I have no idea what it could be...

K900, 2022-11-18:
I've had this happen on a dual stack system where the Pi was only reachable on IPv4, which confused Windows somehow.

pedrolamas, 2022-11-18:
Thank you for reporting that @K900!

@nathanielhudson, can you confirm if you are running a dual stack (IPv4 and IPv6) network?

pedrolamas, 2022-12-24:
FYI, I was unable to repro this issue and so have not confirmed this to be a Fluidd bug.

As such, I have now changed the status of the ticket.

wuspy, 2023-01-18:
I also experience this issue on Firefox on Windows. Chrome/Edge works fine, Firefox also works fine if I access it through the IP instead of the local domain. Mainsail also has the same issue though, so I doubt this is a fluidd bug.

nathanielhudson, 2023-01-23:
> Thank you for reporting that @K900!
> 
> @nathanielhudson, can you confirm if you are running a dual stack (IPv4 and IPv6) network?

I can confirm that this still happens even with IPv6 disabled in windows network adapter properties. Not sure if there's another check I should be doing.

> I also experience this issue on Firefox on Windows. Chrome/Edge works fine, Firefox also works fine if I access it through the IP instead of the local domain. Mainsail also has the same issue though, so I doubt this is a fluidd bug.

This is my experience as well. Might actually be a Firefox bug with websockets and mdns.

warmbo, 2023-03-27:
Also having this issue but it's in every browser. Websocket errors just keep populating and my chances of getting in to Fluidd or even the webcam stream is almost minimal.

pedrolamas, 2023-03-27:
@warmbo any chance you are using IPv6? We've seen some reports of issues with it, just wanted to check if there is anything to it...

Would be great if you can open the Developer Tools (press <kbd>F12</kbd> on the browser) and keep the Console in view so that when it happens again, take a screenshot of any error there and upload it here for us to check.

pedrolamas, 2025-03-07:
This issue seems to have been mitigated through improvements on both Fluidd and Firefox and as far as I can tell, this is not happening currently, so I will be closing this ticket down.

If someone still experiences this issue, I do ask that a new ticket is raised instead.

closed 2025-03-07 by pedrolamas