fluidd-core/fluidd #933: UI crashes when camera fails to load frame

leoshusar, 2022-10-31:
### Fluidd Version

v1.21.1-3a80b5f

### Browser

Chrome

### Device

Desktop PC, Mobile device (Phone, Tablet or similar)

### Operating System

Windows, Android

### What happened

When some webcam frame fails to load, it crashes the whole UI into the "Connecting to Moonraker" screen, see here the network tab:
[image]

It happens everytime exactly after 10 seconds of loading the frame.
I'm using MJPEG Adaptive stream.

### What did you expect to happen

To me it seems like there is a 10 seconds timeout and it crashes the whole UI if the timeout occurs.
I don't know the real reason, but I would say it should just throw the frame away and load next one.

### How to reproduce

Have unstable webcam stream :)

pedrolamas, 2022-11-04:
Thank you for your report @leoshusar.

I haven't been able to reproduce this one, but I wonder if you can check the Developer Console for any error please?

leoshusar, 2022-11-04:
Developer console is empty.
But strange thing I noticed. I tried to set 15 seconds latency using Chrome throttling profile and nothing happened. Then I tried 30 seconds and it still went fine:
[image]
so it looks like the time itself is not the problem.

I also checked request and response headers on successful and crashing request and they are the same.
I'm thinking if maybe the arrived data is not correct and so it crashes?

If you know what else can I do to help, I will try.

pedrolamas, 2022-11-04:
> I'm thinking if maybe the arrived data is not correct and so it crashes?

I did try that myself (by sending corrupt data in a single frame) and while it killed the stream, the whole thing still worked fine!

Something else must be at play here, but without a repro I don't know exactly how to help.

leoshusar, 2022-11-04:
I think I found the problem, but still not the reason.
I noticed when the frame starts failing to load, websockets stop communicating:
[image]
after it fails completely and crashes, it gets all missed messages at once:
[image]

I am running Fluidd behind Cloudflare so I thought CF might be dropping it - but this issue doesn't happen with disabled camera stream (just by using the "hide camera" button). But it also doesn't happen on LAN! Or maybe I haven't tried it for long enough.

I also thought there might be some traffic limit in CF or something, but I have never experienced any issues in my another apps.
I'll see if I can find anything else.