fluidd-core/fluidd #934: Don't auto play Camera and don't auto load Gcode Preview when Mobile Browser is detected

ChaosBlades, 2022-11-01:
### Is your feature request related to a problem? Please describe

I have noticed on Mobile Browsers that the Fluidd WebUI is super slow at best or takes minutes to load on mobile browsers when the Camera and Auto Load of Gcode Preview are enabled. Mobile browsers just can't handle the extremely inefficient mjpeg streams.

Worst case senerio is when I am remote. On my phone that is using LTE internet and I try to load Fluidd. I never loads because the camera is always trying to pull ~20Mbps. Not to mention monthly bandwidth limits.

### Describe the solution you'd like

When a Mobile Browser is detected don't auto play the camera and don't use Auto Load of Gcode Preview even if it is enabled.
Add a setting to enable/disable this feature

kieraneglin, 2023-02-05:
@pedrolamas I suspect this is resolved based on #1028 + per-breakpoint dashboard configs. Thoughts?

pedrolamas, 2023-02-05:
Indeed! I actually thought your PR was based on this issue, but never crossed my mind of actually checking!

I will now close this as complete!

closed 2023-02-05 by pedrolamas