fluidd-core/fluidd #938: Allow disabling of links to configuration documentation in the editor

balthisar, 2022-11-03:
### Is your feature request related to a problem? Please describe

Recently the built-in file editor has started inserting blue “View 'xxx' documentation links”, which cannot be turned off (or at least, the new configuration setting controlling this is not present in the documentation). While it's valuable for some users, for others, they're redundant, visually disturbing, and seriously break up the flow and formatting of text documents.

For example:

[image]

### Describe the solution you'd like

I would like a configuration option — either in one of the config files or in the browser — to turn off this new feature.

### Describe alternatives you've considered

I've considering ssh'ing and using vi for quick and dirty changes, or using BBedit (or another ssh-capable text editor) for file editing, but it's quite convenient to use the built-in editor from time to time.

balthisar, 2022-11-03:
Never mind; apparently this is what "CodeLens" is in the options. At least this closed issue might help others when they search. And I'm an idiot; I simply didn't check before opening this.

closed 2022-11-03 by balthisar

pedrolamas, 2022-11-03:
Glad you found the setting @balthisar, seems we need to improve documentation around this, so there is still a lot of value in this report! 🙂