fluidd-core/fluidd #949: after updating for v1.20.1 fluidd stopped responding, getting toast error: 'Directory does not exist (/home/pi/printer_data/gcodes)'

harellevy, 2022-11-13:
### Fluidd Version

v1.20.1

### Browser

Chrome

### Device

Laptop

### Operating System

macOS

### What happened

after updating last time klipper to 1.20.1 (all shows as up to date),
Just connected today and got a toast error: 'Directory does not exist (/home/pi/printer_data/gcodes)'
and in the HOME i get this warnings:
Moonraker warnings found.
[server]: Option 'enable_debug_logging' is deprecated, see the configuration documention at https://moonraker.readthedocs.io/en/latest/configuration/
Unable to create data path folder at /home/pi/printer_data
[database]: Option 'database_path' is deprecated, see the configuration documention at https://moonraker.readthedocs.io/en/latest/configuration/
Reverting to legacy database path: /home/pi/printer_data/database
[file_manager]: Option 'config_path' is deprecated, see the configuration documention at https://moonraker.readthedocs.io/en/latest/configuration/
[file_manager]: Option 'log_path' is deprecated, see the configuration documention at https://moonraker.readthedocs.io/en/latest/configuration/
Failed to validate service: [Errno 2] No such file or directory: '/home/pi/printer_data'

nothing like uploading files, printing, changing configuration files doesn't work, restart firmware wasn't helping.

### What did you expect to happen

app initial or at least will let me to force version again (reinstall same version) or show new versions to download with fix.

### How to reproduce

install fluid 1.20.1, not in the right order might be the issue, we have too many update buttons, it's better the app will handle all the installation flow, and I would only select what to install.

### Additional information

[image]

harellevy, 2022-11-13:
[image]

pedrolamas, 2022-11-13:
This is not a Fluidd bug, it is actually a know extra step when upgrading to the latest Moonraker.

You will need to follow the instructions shown in the warnings to get your printer back into working condition.

closed 2022-11-21 by pedrolamas