fluidd-core/fluidd #954: Seperate dashboard layouts for desktop en mobile

TheHunter11, 2022-11-17:
### Is your feature request related to a problem? Please describe

The best dashboard layout for the desktop for me isn't the best for when using mobile and vice versa. This has mainly to do with the fact that mobile doesn't show two pannels next to eachother in colomns. This results in panels that are on top in the desktop version in the second colomn end up in the bottom in mobile. Wich is inconvenient do to a lot of scrolling. 

### Describe the solution you'd like

The possibility to make seperate layouts for the mobile version and the desktop version. Im aware of the fact that this feature is available in mainsail, but i like fluidd beter on other points and not worth changing.

pedrolamas, 2022-11-17:
You can do this currently on 1.21! Click on "Adjust dashboard layout" on each size and change accordingly!

[image]

TheHunter11, 2022-11-17:
Im aware of the fact you can change your dashboard layout but this changes it for desktop and mobile at the same time. But you can't make seperate layouts for the desktop and  a seperate one for the mobile version.

pedrolamas, 2022-11-17:
I know this is not exactly "per device", but we now support layout per Display Breakpoints (xs, sm, md, lg) so you should be able to have a layout for desktop (md or lg normally), and mobile (xs or sm normally). Would this be sufficient?

TheHunter11, 2022-11-17:
i found the feature in the release notes. But i don't quite understand how you are supose to do it. Do i assume right that if you change the layout on one device it will stay the same for that device even if you change it on another device? If so i found a problem on the vertical mobile version. When you are in vertical mode you aren't able to exit layout mode thus i assume save that layout.

PR #1331 (2025-03-14): feat: report target layout in layout edit mode
Clarifies the current target layout when in layout edit mode

[image]

Improves #1017 
Resolves #954
closed 2025-03-14 by pedrolamas