fluidd-core/fluidd #960: Wrong HOST temp

kobzzon, 2022-11-22:
### Fluidd Version

v1.21.1

### Browser

Chrome

### Device

Desktop PC

### Operating System

Windows

### What happened

Temp host in Fluidd displaying wrong

### What did you expect to happen

in ssh session host temp is Ok

### How to reproduce

[image]
[image]

pedrolamas, 2022-11-22:
What configuration section are you using to retrieve the host temperature?

Fluidd is just a web-interface and shows whatever Klipper provides (via Moonraker), so this in the end will be a Klipper issue, so not something we can fix here.

pedrolamas, 2022-11-22:
Assuming you are using a `sensor_type: temperature_host` in your Klipper printer.cfg, you will most likely have to override the `sensor_path` as Klipper by default will use `/sys/class/thermal/thermal_zone0/temp` but that doesn't seem to be available in your system.

You can confirm by SSH'ing and then `cat /sys/class/thermal/thermal_zone0/temp` and checking the result

kobzzon, 2022-11-23:
> cat /sys/class/thermal/thermal_zone0/temp

[image]

pedrolamas, 2022-11-23:
I guess that confirms how the -1 appeared!

You will need to find out that the correct path is for your system and set that following the instructions on my post above.

pedrolamas, 2022-11-29:
I am closing this now as I assume you've followed the required steps.

If you are still having trouble with this, I suggest jumping on [Klipper Discord](https://discord.klipper3d.org/) `#configuration` channel and following up on the matter there.

closed 2022-11-29 by pedrolamas