fluidd-core/fluidd #966: Settings to change Jobs sorting order

JacopoBassan, 2022-11-27:
### Is your feature request related to a problem? Please describe

The Jobs panel orders files and folders the same way as the **ls** command, in my use case (and I think most cases) it's very disorienting, for example I have:
```
~/gcode_files $ ls
10-Calibration  11-Upgrades  1-Cosplay  3-Models  4-Toys  5-Keychains  6-Home  7-Rings  8-Vases  9-Fusion360  Ender3
```

### Describe the solution you'd like

I think we should use the -v flag
```
~/gcode_files $ ls -v
1-Cosplay  3-Models  4-Toys  5-Keychains  6-Home  7-Rings  8-Vases  9-Fusion360  10-Calibration  11-Upgrades  Ender3
pi@chonkyraspy:~/gcode_files $ 
```

### Additional information

I prints lots of files that have multiple parts and keep my folders organised and synced so an option to change the sort behaviour would be great. Thanks!

pedrolamas, 2022-11-27:
HI @JacopoBassan , thank you for taking the time to submit this Feature Request.

Can I ask why don't you use 2 or even 3 digits (so instead of having "1, 2, 3..." you would have "001, 002, 003...")? That would surely maintain order as you require here!

JacopoBassan, 2022-11-27:
Hi @pedrolamas, while I do get that I could do that, I'd have to rename more than just a couple of files and folders, I have a lot of files like part1, part2… part10 and I really don't see any use of an ordering that doesn't follow natural language ordering. Plus what's still missing is that at every refresh the UI uses the Modified date for default ordering :)

pedrolamas, 2023-01-18:
Sorting just by numerical prefix is doable, and we are now working on a solution for that, but reproducing the full sort order used in `ls -v` is quite tricky...

To be honest, I haven't been able to find any specifics on the algorithm used!

pedrolamas, 2023-01-18:
Spoke too soon: I found the rules for `ls -v` (and `sort -V`) https://github.com/coreutils/gnulib/blob/master/lib/filevercmp.h

Will now try to implement this...

PR #1013 (2023-01-19): feat: allow sorting text by version or prefix number
Allows user to select the type of sorting to apply on the file list between Default, Number Prefix sort, or Version sort.

[image]

Default is the current implementation, where we just `compare` from `Intl.Collator` for the current locale.

Number Prefix sort will check if the text has a number prefix (digits only) and if so, use that for sorting. When comparing 2 prefixes, if they are the same, then it will fallback to Default sort.

Version sort matches the type of sorting that `ls -v` or `sort -V` would use.

Example of version sorting:

 [... cut]
closed 2023-01-19 by pedrolamas