fluidd-core/fluidd #971: Disturbed menu icons

GP3DS, 2022-12-08:
### Fluidd Version

1.22.0

### Browser

Safari, Other (Please tell us below)

### Device

Desktop PC, Laptop

### Operating System

macOS

### What happened

After updating from 1.21.1 to 1.22.0, the menu icons are no longer visible. The name code of the icons is replaced.

### What did you expect to happen

How to fix this bug?

Is it possible to explain to me how to fix this?

Thank you for your help

### How to reproduce

Just update from 1.21.1 to 1.22.0 (with the update of the operating system packages)

### Additional information

[image]
[image]
[image]
![Capture d’écran 2022-12-08 à 09 54 48
[image]
[image]

](https://user-images.githubusercontent.com/10012680
[image]
5/206402589-73613f17-5dd0-4446-bd7e-8ff535358e2a.png)
![Capture d’écran 2022-12-08 à 09 55 24](https://user-images.githubusercontent.com/1001
[image]
26805/206402629-f76f65e4-246e-4f61-8d9d-e12a68df97d2.png)

pedrolamas, 2022-12-08:
Hi @GP3DS thank you for reporting this.

We've seen other users having the same issue but have yet to determine what is causing this behavior.

Having said that, some users found that they could mitigate this problem by doing a full browser refresh of Fluidd (with <kbd>Ctrl</kbd>+<kbd>F5</kbd>), then going on Fluidd Settings, changing the language to English, and then back to their chosen language and it would then work!

Can you please check if these steps also fixes the problem for you?

GP3DS, 2022-12-08:
Thank you for your quick response.

I've been looking for a solution since yesterday morning... and you solved the problem in 5 minutes ;-)

The fact of moving from French to English and again from English to French solved the problem on my two machines!

Thank you very much for your help!!!

pedrolamas, 2022-12-09:
I believe I found what caused this issue and have just pushed a fix for it in v1.22.1!

closed 2022-12-09 by pedrolamas