fluidd-core/fluidd #980: high amount of dns requests

rebelangel66, 2022-12-15:
### Fluidd Version

1.22.1 and all previous versions over the last 2 years

### Browser

Chrome, Firefox

### Device

Desktop PC, Laptop, Mobile device (Phone, Tablet or similar)

### Operating System

Windows, Linux, Android

### What happened

I'm using a local dns server so Ican give all of my internal services a pretty url instead of having to remember all the ip adresses and ports. If I open the statistics on it the url for my printers are by far the most requested dns adres. It's about 36000 per hour which I'm guessing is the webcam feed refreshing 10 times per second. I've had to up some of the limits on my dns server because it was hitting those limits and causing problems with my internet.

### What did you expect to happen

Can the dns resolution be cached for the webcam feed or just use an ip or localhost or something else.

### How to reproduce

Have an internal dns server and a dns name assigned to fluidd.
Have a webcam feed running.

pedrolamas, 2022-12-15:
Hi @rebelangel66 thank you for reporting this.

I too use a similar setup (AdGuardHome with a forced local DNS entry) so i know exactly what you are discussing here.

We don't control the DNS resolution (that is the browser working), but from your description it seems to me you have a very low TTL set on your A or CNAME entries.

You can indeed try and use an IP for the camera, just go on Fluidd settings and change the camera address there.

rebelangel66, 2022-12-15:
The local dns server sets the ttl of everything to 0 cause it wants to do all the caching itself. It does seem a bit useless to cache everything multiple times if your on the same network. Also they don't want a whitelisted adress to still be blocked in the browsers cache. 
I didn't know you could enter an entire url in the webcam settings but that does solve my problem. Guess this feature is handy if you're using a network camera in stead of a usb cam.

Thanks for the quick reply.

closed 2022-12-15 by rebelangel66