fluidd-core/fluidd #989: Search does not show any results on Job list

idzuwan, 2022-12-20:
### Fluidd Version

v1.22.2

### Browser

Firefox

### Device

Desktop PC

### Operating System

Windows

### What happened

When try to search Job/Uploaded Gcode it will not show any result
[image]
[image]

same result on chrome or mobile

### What did you expect to happen

able to search uploaded gcode

### How to reproduce

using search function on the job list

### Additional information

I noticed it only started to happen but didnt remember exact day when

pedrolamas, 2022-12-20:
Thank you for the report @idzuwan.

For the record, we've been aware of this issue for a while now, it is a [bug in Vuetify library](https://github.com/vuetifyjs/vuetify/issues/11600) we use and only became a problem for us with the changes from #812.

idzuwan, 2022-12-20:
Hi @pedrolamas 

Thank you, it good to know that it is been work on at the moment

Rgds
Idzuwan

PR #992 (2022-12-23): fix: mitigates Vuetify bug filtering items ahead
The source of this problem can be traced to this [open issue in the Vuetify library](https://github.com/vuetifyjs/vuetify/issues/11600), but this only became an issue for us with the changes from #812.

This fix refactors the code to not rely on the Vuetify DataTable filter capability and instead we now filter the items ahead of binding to the DataTable

Fixes #989
closed 2022-12-23 by pedrolamas