fluidd-core/fluidd #1000: Make AppBar upload & print icon consistent with the icon in the jobs section

leoshusar, 2023-01-02:
### Is your feature request related to a problem? Please describe

PR #974 added upload & print icon to the AppBar:
[image]

But in the jobs section this icon means only upload:
[image]

### Describe the solution you'd like

I think it would be good to make these icons consistent, probably change the AppBar icon to be the same as the other one

pedrolamas, 2023-01-03:
Thanks @leoshusar, you make a very valid point there (I too value consistency as much as possible)!

But having said that, I'm not sure I like this icon on top...

[image]

leoshusar, 2023-01-03:
Hmm, I don't have too much sense for design so I think I won't be helpful here :) I personally would be okay with this icon, but we can see what others think about it.

What about finding completely another icon for upload & print?
Maybe `publish`? Would it make sense here?

[image]

[image]

[image]

PR #1004 (2023-01-03): style: make upload & print icon consistent
Use correct icon for "Upload and Print" functionality (using same one as in the popup menu of the Files card).

[image]

Fixes #1000