fluidd-core/fluidd #1003: Cannot cancel GCode preview loading when using "Automatically load file on print start"

leoshusar, 2023-01-03:
### Fluidd Version

1.22.2

### Browser

Chrome

### Device

Desktop PC, Mobile device (Phone, Tablet or similar)

### Operating System

Windows, Android

### What happened

I am using GCode preview and I have enabled "Automatically load file on print start".
Sometimes when I'm checking from my phone using mobile data, I want to cancel that preview download, because my data plan has limited speed and I only want to quickly check the webcam. However it's currently impossible to cancel it, the dialog just reappears and the download continues (with NaN as remaining download size).

https://user-images.githubusercontent.com/13891066/210394523-045dd0b8-5121-4aa7-a83d-59456dbe36c8.mp4

### What did you expect to happen

Downloading should stop

### How to reproduce

- Enable "Automatically load file on print start"
- Open Fluidd
- Try to cancel the preview download (you can use Chrome network throttling for better reproduction)

Matrixchung, 2023-01-13:
The same issue even occurs when "Automatically load file on print start" is disabled and all "Gcode preview" settings are default. After clicked the "cancel" button, the download dialog just reappears with a total file size of "NaN KB".

pedrolamas, 2023-01-13:
Thanks, I've managed to reproduce and confirm this bug. We will try to fix this on the next Fluidd release.

PR #1033 (2023-02-11): feat: make file download dialog component independent from parent
@pedrolamas informed me of an issue where the file download dialog wouldn't appear if the `Jobs` (or, really, the `FileSystem`) component wasn't actively rendered on-page.

This PR refactors the way the file download dialog is interacted with by making it independent of any specific parent component. To accomplish this, I also had to refactor the way Axios `cancelTokenSource`'s were created/referenced by adding them into Vuex.

Fixes #1003
closed 2023-02-11 by matmen