fluidd-core/fluidd #1010: fluidd UI keeps resetting randomly

petemagnusson, 2023-01-12:
Hi, 
Every now and again my fluidd installation keeps resetting the UI.
Layout
Macros (categories) 
Theme
Thermal Presets.

Everything that is in the cfg and txt files are of course untouched.

Not a major problem but just a bit of a pain to setup again :)

I have tried different mSD cards and also two different pis ( 3 initially and now a 4 ).

I have not really looked at the logs but happy to share if you could point me to which one would be helpful.

Thanks and i really appreciator the work and usefulness of fluidd so huge thanks for all your work!

Arakon, 2023-01-13:
I can confirm this issue, it has happened multiple times to me on two different PCs, both with firefox, across the past years (and versions, obviously), and across two different printers.
Running on Raspberry Pi 3B and Raspberry Zero 2 W.

pedrolamas, 2023-04-14:
I have no idea if your issue is related, but if you are connecting via WiFi, you might want to try installing [Sonar](https://github.com/mainsail-crew/sonar) and see if that helps with the problem.

pedrolamas, 2023-12-01:
I believe this has been fixed in the most recent Fluidd releases, so closing the ticket as such.

closed 2023-12-01 by pedrolamas