fluidd-core/fluidd #1018: Instant (sort of live) speed update for tool head movement

dhruvinsh, 2023-01-26:
### Is your feature request related to a problem? Please describe

NA

### Describe the solution you'd like

We have filament flow Measurement but missing the speed for tool head movement. 

While we have co-ordinates updates present, Pedro suggested over discord that its easy to display on dashboard.

pedrolamas, 2023-01-27:
@dhruvinsh thanks for creating this Feature Request!

Would something like this work for you?

[image]

Using the same style as in the live coordinates, we would just add the live speed to the "Speed" label, as shown on the above image

dhruvinsh, 2023-01-27:
@pedrolamas  that look really nice. Easy to read and placement also looks good. 

Although, having the tool head speed and flow rate together, would that make sense? Just an opinion. Else current implementation looks good.

PR #1019 (2023-01-29): feat: show live speed while printing
Adds live speed to the Tool panel

[image]

Using the same style as in the live coordinates, we would just add the live speed to the "Speed" label, as shown on the above image.

Fixes #1018
closed 2023-01-29 by pedrolamas