fluidd-core/fluidd #1030: Power toggle in top navigation not working

dhruvinsh, 2023-02-05:
### Fluidd Version

v1.23.1

### Browser

Chrome

### Device

Desktop PC

### Operating System

Windows

### What happened

Hope I am not misunderstanding the usage of the option but from `Settings-> General->Power toggle in top navigation` can not be set
[image]

Klipper and Moonraker options are not clickable. 
 

### What did you expect to happen

I would expect that restart button for klipper and moonraker in the top navigation bar.

### How to reproduce

I haven't try the previous build but with current release it should be there. 

### Additional information

Not sure but this commit e71222873546dc2736051ed51ddfc21c01176c13 might be fixing it. I haven't got the chance to look at it. Made a judgment based on code only.

pedrolamas, 2023-02-05:
Power toggle will show a button for either a Klipper [output pin](https://www.klipper3d.org/G-Codes.html#output_pin) (if any set on printer.cfg) or a Moonraker [power device](https://moonraker.readthedocs.io/en/latest/configuration/#power) (if any set on moonraker.conf)

The above list is empty as you have neither, so you can't select any.

We will make a quick change on the UI to hide the "Klipper" and "Moonraker" headers when no entries are available to avoid further confusion!

dhruvinsh, 2023-02-05:
Ahh make sense. To begin with my understanding was incorrect. Thank you for confirming. I will close this issue now.

closed 2023-02-05 by dhruvinsh