fluidd-core/fluidd #1032: Unable to set tool extrude length > 50mm

thijstriemstra, 2023-02-08:
### Fluidd Version

1.23.1

### Browser

Firefox

### Device

Desktop PC

### Operating System

Windows

### What happened

In the system settings it's possible to change the default extrude length to a 100mm.

[image]

But this screws up the tool interface on the dashboard and it seems like it won't allow higher than 50 there?

[image]

### What did you expect to happen

Being able to use > 50mm extrude length

### How to reproduce

Try to set to > 50 extrude length in tool interface on dashboard.

thijstriemstra, 2023-02-08:
I see, thanks @pedrolamas! In that case the system settings panel shouldn't allow more than `max_extrude_only_distance` right (and right now it does)?

pedrolamas, 2023-02-08:
That's a very fair point, and yes, I believe there is a bug there (even if we allow to enter a setting over the Klipper limit, the interface should not use that).

PR #1038 (2023-02-14): feat: updated tool extrusion settings to respect printer config
Fixes #1032

Updates the `ToolheadSettings` component to respect an extruder's `max_extrude_only_velocity` and `max_extrude_only_distance` when trying to update the default extrusion distance/speed values. Also adds a last-resort speed/distance check on extrusion/retraction commands
closed 2023-02-14 by pedrolamas