fluidd-core/fluidd #1044: Slider input lockout not present on all tablets

kieraneglin, 2023-02-21:
### Fluidd Version

Latest develop

### Browser

Safari, Other (Please tell us below)

### Device

Mobile device (Phone, Tablet or similar)

### Operating System

Android, iOS

### What happened

Slider input locks aren't present on tablets that are larger than the Vuetify mobile breakpoint 

### What did you expect to happen

Slider input locks should be present on tablets. Possible solutions:

- add checks for touch screens - possibly in tandem with a larger breakpoint cutoff to not show locks on touchscreen laptops. ([relevant SO question](https://stackoverflow.com/questions/56324813/how-to-detect-touch-device-in-2019))
- change the cutoff to include larger devices. This has bigger ramifications as there are a lot of independent definitions of `isMobile` in the codebase that would all need their tablet behavior considered. 
  - tangential: it may be worth refactoring `isMobile` into a helper method that's then imported and called by the views. DRY it up and all that

### How to reproduce

Open Fluidd UI on a normal tablet (an iPad, for example)

kieraneglin, 2023-02-21:
@pedrolamas if there is consensus on how to handle this, I can open a pull request with a fix!

pedrolamas, 2023-02-21:
I don't use tablets, so I never came across this... having said that, we do have a better `is-mobile` method that checks for browser agent: https://github.com/fluidd-core/fluidd/blob/develop/src/util/is-mobile.ts

At the moment, we only use it to ensure we don't show Monaco editor on mobile devices as it is not supported and most likely wouldn't work.

My guess is that this method would be a better fit for purpose than using the existing Vuetify check!