fluidd-core/fluidd #1049: Save Z Offset without restarting klipper

simpat1zq, 2023-02-28:
### Is your feature request related to a problem? Please describe

Not really a problem, but I don't think there's really any need to restart klipper when saving the Z Offset. But it would be nice to be able to save it without the firmware restarting. 

### Describe the solution you'd like

Have a choice between save z offset, and save+restart.

pedrolamas, 2023-03-03:
Thanks for opening this @simpat1zq.

Currently, there is no way to save the z-offset without restarting Klipper, that is just the way it currently works!

The "live" z-offset will work fine, but to actually have it saved, Klipper requires that `SAVE_CONFIG` command is called, and that will always do a firmware restart.

Feel free to open a Feature Request on the Klipper repo, as this is not anything that Fluidd can do without changes on Klipper side.

pedrolamas, 2023-04-19:
The changes in #1086 have removed the automatic call to `SAVE_CONFIG`, so now the user will have to manually call it, or just click "Save Config & Restart" button on Fluidd.

closed 2023-04-19 by pedrolamas

Stooovie, 2023-09-06:
Alas, that doesn't work when the user has a script to turn the printer off after printing for example. It sucks for remote control.