fluidd-core/fluidd #1053: Sensor types failing to be acknowledged.

BlackNet, 2023-03-03:
### Fluidd Version

latest

### Browser

Chrome, Firefox, Safari, Edge, Other (Please tell us below)

### Device

Desktop PC, Laptop, Mobile device (Phone, Tablet or similar), Other (Please tell us below)

### Operating System

Windows, macOS, Linux, Android, iOS, Other (Please tell us below)

### What happened

Any sensor listings with field names NOT listed in the below table are ignored and not displayed.

`export interface Sensor {
  name: string;
  prettyName: string;
  key: string;
  color?: string;
  type: string;
  temperature: number;
  pressure?: number;
  humidity?: number;
  target?: number;
  measured_min_temp?: number;
  measured_max_temp?: number;
  maxTemp?: number;
  minTemp?: number;
}`

### What did you expect to happen

To display all sensor data.

### How to reproduce

Use of any sensor that uses values other than temp, humidity, or pressure, i.e., bmeXXX, sgp30, sgp40 (gas/voc/raw/altitude)

### Additional information

Values being read from klipper/moonraker.

Values for bme860 sensor being excluded are 'gas', 'altitude' is not included in the python script.
`{"result": {"eventtime": 101907.375592499, "status": {"bme280 BME680": {"temperature": 22.52, "pressure": 995.3988310515684, "humidity": 61.29085867958693, "gas": 12858837.485172005}}}}`

Values for sgp40 sensor being excluded are 'gas', 'voc',  'compensated_raw_gas', 'raw',  is not included in the python script.
`{"result": {"eventtime": 102016.242367304, "status": {"sgp40 SGP40": {"temperature": 22.5, "humidity": 61.45285423930575, "gas": 33125, "voc": 0}}}}`

pedrolamas, 2023-03-03:
Thank you for opening this issue, @BlackNet.

This is not actually a Bug (more of a missing feature), and as such I've changed the type to the more appropriate Feature Request.

BlackNet, 2023-03-03:
I would actually classify it as a dual issue, one part bug and one part unexpected behavior for supported values from klipper. I am attempting to add sensors like sgp40 to klipper and all related modules when I ran into this one.

pedrolamas, 2023-03-07:
@BlackNet We can add the "gas" component from BME680, but unless I am missing something, SGP40 is currently not supported by Klipper, so until it gets added there, there is nothing we can do on Fluidd side.

BlackNet, 2023-03-07:
That would be wonderful. I am working with some code for the sgp40 and I think it would also get picked up correctly if gas or voc was listed for the bme class sensors.