fluidd-core/fluidd #1058: Finish date-time in the PRINTING section does not respect the configured date format

Anathema-Device, 2023-03-07:
### Fluidd Version

current

### Browser

Firefox, Other (Please tell us below)

### Device

Desktop PC

### Operating System

Windows, macOS

### What happened

I have both the date and time formats set to ISO8601.  Today is 2023.03.07 the finish time shown for the current print is displayed as "08/03 01:48" which I would interpret as MM/DD HH:MM.  In the case of ISO8601 I would expect it to be "03/08 01:48"  (tomorrow morning at 1:48 AM).  So it looks like the Finish time is being displayed using the locale time format not the configured time format, if that makes sense.

### What did you expect to happen

I expect to see the Finish time formatted according to the date and time formats chosen in the fluidd settings.  In this case as MM/DD HH:MM.

### How to reproduce

Configure the date & time formats to be ISO8601 then start a print and observe the Finish time in the PRINTING section

### Additional information

Thanks for the great work, I really enjoy fluidd !

Anathema-Device, 2023-03-09:
Thank you for looking at this. Maybe it is a bug in Javascript?   When I look at the 8061 stuff it sez "the order of the elements used to express date and time in ISO 8601 is as follows: year, month, day, hour, minutes, seconds, and milliseconds."  So it does not say that if the year is not presented then the order would change.  It would not be terrible to present the year there too as it is conceivable that a print started at the end of one year would finish in the next.  Very small but interesting problem though ;)

closed 2023-03-09 by pedrolamas

pedrolamas, 2023-03-09:
> Maybe it is a bug in Javascript?

Not actually a bug as `Date.prototype.toLocaleDateString()` just doesn't have any support for ISO-8601, only real locales...

We were trying to "fake it" by using "sv-SE" as the closest locale matching ISO-8601... till you notice that it also failed on that particular case! 😁

It seems using "lt" locale is a closer match and fixes the issue you pointed out, so we will be using that one going forward!

PR #1061 (2023-03-09): fix: use Lithuanian "lt" for ISO-8601 formats
As there is no way to use `Date.prototype.toLocaleDateString()` with ISO-8601, we have been using "sv-SE", but apparently this doesn't not handle correctly when we ask to omit the year.

I've checked all the locales I could find and "lt" (Lithuania) matches all the cases I have tested, so let's use that one going forward!

Fixes #1058