fluidd-core/fluidd #1063: Multiple printers configured but the same names are displayed

morel666, 2023-03-10:
### Fluidd Version

1.23.1-8b3b58

### Browser

Chrome, Safari

### Device

Desktop PC, Mobile device (Phone, Tablet or similar)

### Operating System

Windows, iOS

### What happened

I have 2 printers configured. In a printer selection pane ("three-dots button" on the top right) both printers have the same name which seems to be inherited directly from Settings -> General -> Printer Name. When I change the name of the one printer, the name of the other printer automatically changes to the same as the first. Opposite way - the same - when I change the name of the second printer (Settings -> General -> Printer name after switching to the second printer) names of both printers are updated to the name of the second one.

### What did you expect to happen

Each printer will have an individual name displayed.

### How to reproduce

1. configure 2 printers (add first printer, add second printer to fluidd)
2. change the name of first printer via Settings -> General -> Printer Name
3. check names of printers in the printer selection pane

### Additional information

Both printers are configured physically on one Raspberry PI. Each printer has it's own (separate configuration files, ports, paths, processes) klipper/klippy, and moonraker instance configured and running,

pedrolamas, 2023-03-11:
Hi @morel666, thank you for reporting this, however from my tests, this feature seems to be working as expected.

I've configured two printer instances (each with its own Klipper and Moonraker instances, using port 7125 for printer 1, and 7126 for printer 2), added each one to Fluidd.

Then I entered printer 1 and changed the name to "fluidd1" on the Settings, selected printer 2 and changed the name to "fluidd2" on the settings, and now have different name, no issue at all!

The only way I see you having this issue is if both moonraker instances are using the same database file!

morel666, 2023-03-11:
Hi @pedrolamas, thanks for the prompt response. You are right - I made a mistake when configuring moonraker and both instances are run with the same -d param value. Thank you for the hint, I thought I had checked everything but it looks like I missed this one.

closed 2023-03-11 by morel666

pedrolamas, 2023-03-11:
Not a problem, glad you got it fixed in the end! 🙂