fluidd-core/fluidd #1068: [gcode preview] failure to parse gcode sliced with Cura 5.3.0

chelli, 2023-03-16:
### Fluidd Version

1.23.3

### Browser

Chrome

### Device

Desktop PC, Laptop

### Operating System

Linux

### What happened

Since switching from Cura 5.2.2 to the most recent stable release 5.3.0, fluidd gcode preview fails to parse the gcode.

See the attached gcode files as an example:
[FQ5_square.zip](https://github.com/fluidd-core/fluidd/files/10995391/FQ5_square.zip)

When trying to load the file, fluidd shows the following and it does not proceed anymore:
[image]

### What did you expect to happen

Fluidd should properly load the gcode files into the gcode preview.

### How to reproduce

Please try to load the attaced (zipped) gcode file into the gcode preview.

pedrolamas, 2023-03-16:
Thanks for reporting this @chelli, I have managed to repro this issue, although it seems more related with the `preprocess_cancellation` script output (if enabled on moonraker.conf, this is run automatically on g-code file upload)

Here we can see an `EXCLUDE_OBJECT_DEFINE` entry without any `POLYGON` or even `CENTER` attributes.

[image]

And this is what Fluidd shows when printing this file:

[image]

I'm not sure what this "NOMESH" object is or how it got into your gcode... could this be some plugin you are using in Cura?

pedrolamas, 2023-03-16:
I've found this reference on Cura: https://github.com/Ultimaker/Cura/issues/14679#issuecomment-1446307269

Seems to be a typo in Cura 5.3 Beta 1, so I would expect this to have been fixed on the final Cura 5.3 release.

pedrolamas, 2023-03-16:
Here's the open PR on Cura repo to fix this issue: https://github.com/Ultimaker/CuraEngine/pull/1855

PR #1069 (2023-03-16): fix: only add parts with defined polygon
Ideally, all `EXCLUDE_OBJECT_DEFINE` have a `POLYGON` parameter, but a bug in Cura has caused this to not be the case.

This fix ensures we only attempt to show parts with a defined polygon on the G-code Previewer.

Fixes #1068
closed 2023-03-16 by pedrolamas