fluidd-core/fluidd #1081: I loose connection every hour or so while printing

supergarixd, 2023-04-10:
### Fluidd Version

latest 1.23.4

### Browser

Other (Please tell us below)

### Device

Other (Please tell us below)

### Operating System

Linux

### What happened

When i connect to the fluidd interface trough its ip after some time (a hour or somenting like that) i get:
[image]
(connecting with moonraker)
after some seconds it disconnects from the wifi.

### What did you expect to happen

That my fluidd interface doesn't disconnect

### How to reproduce

I was on the earlier version of fluidd and everything worked fine but i updated and this starts to happen.

### Additional information

My hardware is a raspberry pi zero W.
The raspberry pi zero is very close to the wifi.
Browser: Opera
[klippy (1).log](https://github.com/fluidd-core/fluidd/files/11193827/klippy.1.log)
[moonraker.log](https://github.com/fluidd-core/fluidd/files/11193828/moonraker.log)

pedrolamas, 2023-04-13:
Hi @supergarixd, thank you for reporting this.

Do you know what Fluidd version were you previously using? Do you have a webcam connected to the Raspberry Pi Zero W?

I have a feeling this might be the same issue we have reported on #931

pedrolamas, 2023-04-13:
I have no idea if your issue is related, but if you are connecting via WiFi, you might want to try installing [Sonar](https://github.com/mainsail-crew/sonar).

Just follow the "Install" instructions on that page and test!

supergarixd, 2023-04-19:
Yeah i did install Sonar and that makes it keep printing even if it disconnects, however when it disconnects i cant change settings, cancel prints or use the interface.
I am not using a webcam.
I think the earlier version of Fluidd was 1.23.3

pedrolamas, 2025-03-07:
This issue seems to have been mitigated through improvements on Fluidd and as far as I can tell, this is not happening currently, so I will be closing this ticket down.

For the record, most issues around WiFi disconnection disappear completely when Sonar is installed!

If someone still experiences this issue, I do ask that a new ticket is raised instead.

closed 2025-03-07 by pedrolamas