fluidd-core/fluidd #1084: Adjusting logic for applying gcode offset into printer.cfg when both enstop and probe are used (no virtual pin)

Endracion, 2023-04-16:
### Fluidd Version

v1.23.4

### Browser

Chrome

### Device

Desktop PC

### Operating System

Linux

### What happened

Noticed after chatting in the Klipper discord that if you have both an active z endstop (used for homing) and a probe, in my case a clicky switch, that the gcode z offset in the fluidd UI used for babystepping while the mesh is loaded actually applies to the z endstop, and not to the z offset like the text would suggest.

[image]

(before)
[image]
(after)
[image]

A user identified this behavior in the code: https://github.com/fluidd-core/fluidd/blob/13e7394ba37f3cc2c79f530c3b66324a12f06260/src/components/widgets/toolhead/ZHeightAdjust.vue#L133-L142

The suggestion mentioned was that if a mesh is loaded, the z gcode offset when saved to the cfg should apply to the z offset and not to the endstop, because right now if I save the config through the prompt, I have to yet again adjust the offset when it restarts, it's like it doesn't hold it in place (and I'm not sure what the logic behind that is anyway).

### What did you expect to happen

For the z offset to be applied to the z offset, not to the endstop.

### How to reproduce

Described above too, but basically have an active z endstop and a probe defined, no so virtual pins.

When you use the UI's z offset to change it on the fly then save it to config, it saves it to z endstop instead of z offset.

pedrolamas, 2023-04-16:
Thanks for opening this issue @Endracion, as I said on Discord I am not internally sure of what the best  behavior would it be for this, but I do understand your point!

We will try to get further feedback on this and see how we can improve this functionality for a next release of Fluidd.

Endracion, 2023-04-16:
As an additional comment, I can't think of any situation where you would want to change the z endstop if you have a probe defined. So instead of checking for the virtual pin, maybe just look at if probe is defined, and if so, z offset should be applied.

pedrolamas, 2023-04-18:
FYI, I've taken a look at this and it seems to be at odds with #546 and the changes introduced in #547

Not sure which behavior is "more" correct/expected...

Endracion, 2023-04-18:
Mainsail's implementation does away with logic, since clearly there are more considerations, and just gives you a dropdown so you get to decide:
[image]

pedrolamas, 2023-04-18:
Now that might actually be the best solution here!! I will look into implementing something similar!

PR #1086 (2023-04-19): fix: allow applying z-offset to endstop or probe
Instead of checking for type of probe/endstop, we will now check for the existence of `Z_OFFSET_APPLY_ENDSTOP` and `Z_OFFSET_APPLY_PROBE`

If only one of these exists, then we save directly via that command; if both exist, we will show a menu and allow the user to pick which to use.

We are also NOT forcing the restart, and instead letting the user decide if/when to do it via the "Save Config & Restart" button on top of Fluidd.

## Single case

[image]

## Multiple case

[image]

Fixes #1084