fluidd-core/fluidd #1087: I updated from 1.19 to 1.23.4 and now my UI is oddly scaled

lDavidl, 2023-04-19:
### Fluidd Version

1.23.4

### Browser

Chrome, Edge, Other (Please tell us below)

### Device

Desktop PC

### Operating System

Windows

### What happened

I just updated everything (been a while) and now my dashboard looks squished. I came from 1.19 and updated to 1.23.4.

I've added some screenshots of how it was before and after I updated. I just happened to take those screenshots coincidentally. 

### What did you expect to happen

Well... Certainly not this.

### How to reproduce

I would assume that you might be able to reproduce this bug if you update from 1.19 to 1.23.4. I however cannot be sure of that.

### Additional information

[image]
[image]
[image]
[image]
![2023-4-19 20-22-55](https://user-images.githubusercontent.com/42553374/233181118
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
-5543086f-7ba0-41a8-b2d1-75a878f5a4ea.png)

pedrolamas, 2023-04-19:
Layout was actually broken in the version you had!

The layout you now see is how it looked before the version you had (and matches what is still in use)

Having said that, you can add a 3rd or even 4th column on the main page, so you can take full advantage of your large screen!

lDavidl, 2023-04-19:
Thank you for the super quick reply! :)

That is a bit surprising to hear as I had been using Fluidd since mid 2021.

And I admittedly don’t really like change so I can’t imagine simply not having noticed a UI change like this one.

Either way, is there a chance to get the “broken” UI restored? I genuinely think that it looks more appropriately scaled. I don’t really have a need for a third column (already tried) and would prefer the larger columns.

While I could just install 1.19 on a new SD Card, I would not love to not have all of my stats (History, Time, ...).

Thanks again for your time!

pedrolamas, 2023-04-19:
I'll leave this as a feature request, it's not on top of my priority list but maybe someone else might pick and work on it!

lDavidl, 2023-04-20:
Thank you, I really hope so!