fluidd-core/fluidd #1089: TMC2240: show temperature in graph

NyxCode, 2023-04-24:
### Is your feature request related to a problem? Please describe

Some TMC drivers, like the tmc2240, have an inbuilt temperature sensor.  
Currently, one has to define a macro to read the `printer.extruder.temperature` variable manually.

### Describe the solution you'd like

It woud be great if this temperature could be seen in the graph along the other sensors.  

### Describe alternatives you've considered

I am suprised that klipper did not chose to expose this temperature reading as a thermistor, which would have automatically resulted in support from mainsail/fluidd. [see here](https://github.com/Klipper3d/klipper/pull/6145).

NyxCode, 2023-04-25:
I have a TMC2240 & would be happy to do some testing.

NyxCode, 2023-04-25:
One complication is that the temperature is only available if the driver is active. Otherwise, `printer["tmc2240 extruder"].temperature` is `None`.

NyxCode, 2023-04-25:
Hm, so I'm new to this, but getting all steppers seems difficult. We can get the names of the steppers through `printer.stepper_enable`, but that doesn't tell us the driver used. I suppose we could try reading ``printer[`tmc2240 ${stepper}`]`` for every stepper we find in `printer.stepper_enable`.

pedrolamas, 2023-04-25:
In theory, all you should need to do is adding 'tmc2240' to this array:

https://github.com/fluidd-core/fluidd/blob/f0986e4b70719f4e154e3fef6ad8f4052a696c31/src/store/printer/getters.ts#L605-L609

closed 2023-08-04 by pedrolamas