fluidd-core/fluidd #1091: Webui doesn't load on an old browser

LAZER555, 2023-04-27:
### Fluidd Version

v1.23.3

### Browser

Chrome, Firefox

### Device

Mobile device (Phone, Tablet or similar)

### Operating System

Android

### What happened

I'm trying to connect to fluidd webui on old android tablets and phone (4.4 android), but i have only whitescreen. On another phone on Android 10 it was the same until i updated the chrome to the latest version. So the question is, do you need the latest chrome version for webui to work? I just wanted to reuse my old tablet as a printer screen.

### What did you expect to happen

Connection to webui

### How to reproduce

Launch on an old 4.4 android phone

pedrolamas, 2023-04-27:
That's expected behavior as Fluidd does require a modern (recent) browser to work correctly.

I'd recommend you either update your browser or if possible, try with another (Firefox or Edge for example)

closed 2023-04-27 by pedrolamas

LAZER555, 2023-04-27:
> That's expected behavior as Fluidd does require a modern (recent) browser to work correctly.
> 
> I'd recommend you either update your browser or if possible, try with another (Firefox or Edge for example)

Well you can't update the browser on 4.4 android. Does mainsail requires latest browser too?

pedrolamas, 2023-04-27:
I believe their minimum requirement is currently the same as Fluidd (both apps use the same components under the hood), but it will be a matter for you to try on your end and see what happens.