fluidd-core/fluidd #1092: Add an Option to open setting files folded

XoBoR, 2023-04-28:
### Is your feature request related to a problem? Please describe

When I open a .cfg file in the Editor, i almost always instinctively hit "fold all" to navigate the file and get a overview. In case anybody doesn't know this very usefull feature, it gets you from this: 
[image]

to this: 
[image]

### Describe the solution you'd like

I would like a little Option in the File Editor section to open cfg files with all sections folded.

### Describe alternatives you've considered

Alternatively an even better approach would be to store the "fold state" between multiple openings of the same file, but that is probably considerably more effort.

pedrolamas, 2023-05-07:
Hi @XoBoR thank you for opening this Feature Request!

So far, I haven't found a proper way to force a file to open with ranges collapsed...

However, I am looking into storing the current state of the regions, so when one opens the same file again, it will have the same collapsed regions!

I still need to evaluate this further, but I believe this might be a good compromise for the problem, so you will be able to manually call "collapse all regions" and the editor would remember that the next time you opened the same file.

XoBoR, 2023-05-07:
> Hi @XoBoR thank you for opening this Feature Request!
>
> So far, I haven't found a proper way to force a file to open with ranges collapsed...
>
> However, I am looking into storing the current state of [...]

Hi @pedrolamas thanks a lot for looking into it! Storing the collapsed state of the regions indeed sounds like the best solution, and it is  what most modern IDEs do too I think. I just thought that blindly collapsing everything could be "good" enough and seemed to be either to implement.

Thanks again for everything, cheers!

PR #1099 (2023-05-11): feat: save and restore editor view state
This PR adds a new Monaco Editor option to "Save and restore view state" (defaults to enabled)

With this option enabled, any file that the user opens will have the current view state (scroll position, selected areas, folds/unfolded regions, ...) stored in the browser local storage when the user closes the editor, and restored when the user re-opens the file.

https://github.com/fluidd-core/fluidd/assets/85504/872b5f70-4632-47bc-ace3-6ac30b9f85a6

To ensure there are no collisions between multi-instances, roots, paths, and filenames, this is the process [... cut]
closed 2023-05-11 by pedrolamas