fluidd-core/fluidd #1104: Exclude object locks the printer when there are diacritics in the name

Fifsson, 2023-05-20:
### Fluidd Version

v1.24.0

### Browser

Firefox

### Device

Desktop PC

### Operating System

Windows

### What happened

When I started a print with and object that has polish diacritics like "ę" in the object name in gcode (the exclude object tag thingy), the printer "paused" as soon as it got to that line of gcode. The "resume" button appeared, but after clicking it the printer responded with an error "Print is not paused, resume aborted". It responded to commands like homing, moving etc. but I was unable to select and print any other file (the "cancel" button also did not work), the only way was to restart it.

I did some testing to confirm that this is the cause and I believe I'm right - manually changing the characters in the object name in the gcode editor resolved the issue.

### What did you expect to happen

I expected the print to start as usual.

### How to reproduce

-create a gcode file with "EXCLUDE_OBJECT_DEFINE NAME=ę" inside (tho I think any not-ascii character is gonna work) and maybe for example G28 to see that it's not responding to the gcode

### Additional information

How it looks in the web interface
[image]

pedrolamas, 2023-05-21:
Hi @Fifsson thank you for reporting this.

Can you confirm what Python version are you using with Klipper and Moonraker? I ask this as Python 2 does not support any non-ascii chars, so that might be at play here.

Fifsson, 2023-05-21:
I'm not shure - "python --version" returned 2.7.16, but "python3 --version" returned 3.7.3. If that's important, I migrated to Fluidd in early 2022.

pedrolamas, 2023-05-21:
Run the following command: `ls ~/klippy-env/bin/python*`

That will show you what Python version is Klipper using.

For Moonraker, use `ls ~/moonraker-env/bin/python*`

--

On a side note, I have just tested with that exact diacritic and it worked fine on my test!

[image]

Fifsson, 2023-05-21:
That's what i've got. 
[image]
So yeah, it's probably that. I guess I'm just gonna avoid the signs in my filenames.

pedrolamas, 2023-05-21:
My advice is that you backup your config, and then start over with Raspberry Pi OS Lite 32-bit and use Kiauh to install Klipper, Moonraker, and Fluidd.

closed 2023-05-21 by pedrolamas