fluidd-core/fluidd #1105: TMC Drivers: Report OTPW

NyxCode, 2023-05-28:
### Is your feature request related to a problem? Please describe

Most (or all?) TMC drivers report OTPW (over-temperature pre warning) conditions. This happens before actual over-temperature occurs & the driver shut down.
This information can be queried through TMC registers. All drivers I'm aware of report the `otpw` flag in `DRV_STATUS`, which can be read through `DUMP_TMC STEPPER={stepper} REGISTER=DRV_STATUS`. The result of this will look something like this is an overtemperature pre-warning is raised:
```
DRV_STATUS: 84004008 sg_result=8 stealth=1 otpw=1(OvertempWarning!) stst=1
```

Currently, users are unaware of these warnings, unless the pre-warning turns into an overtemperature condition, and their printer shuts down.

### Describe the solution you'd like

It'd be great if this information could be shown in the UI.  
I think it'd probably fit best in the notifications, where e.g a Raspberry Pi overtemperature warning would be shown.  

In certain cases, it might be desirable to ignore these warnings - for example, if the user made sure that their drivers don't heat up past the overtemperature pre-warning.

NyxCode, 2023-05-28:
> In certain cases, it might be desirable to ignore these warnings - for example, if the user made sure that their drivers don't heat up past the overtemperature pre-warning.

This might beout-of-scope for fluidd, since all drivers I'm aware of allow users to configure the threshold for these values.  
For example, from the TMC2209 datasheet:
[image]

pedrolamas, 2023-08-05:
Hi @NyxCode, thanks for opening this Feature Request.

I've taken a look at this and though we could manually query the driver, that is not the correct way; ideally, Klipper should provide this information as part of the `TMCxxxx` driver status object, which it currently does.

If and when that situation changes, we are willing to revisit this and consider adding the requested support.

pedrolamas, 2024-06-05:
I might have jumped the gun here... I just checked and it seems that the `TMCxxxx` status object is a "live" status, so we can use that to check for "otpw" (if it's not there = all good, if it appears = warn user!)

pedrolamas, 2024-06-10:
> I might have jumped the gun here... I just checked and it seems that the `TMCxxxx` status object is a "live" status, so we can use that to check for "otpw" (if it's not there = all good, if it appears = warn user!)

I have received confirmation that this is indeed possible with the current TMC driver status, so this is now going into the backlog for future implementation.

PR #1448 (2024-06-12): feat: show warning for Trinamic OTPW flag
Shows a notification message when any `TMCxxxx` driver reports an `otpw` flag (over-temperature pre warning)

[image]

The link on the notification is currently pointing to https://www.klipper3d.org/TMC_Drivers.html#tmc-reports-error-ot1overtemperror

Resolves #1105
closed 2024-06-12 by pedrolamas