fluidd-core/fluidd #1113: Export the configuration file of Fluidd for easy recovery when reinstalling

gaifeng8864, 2023-06-21:
### Is your feature request related to a problem? Please describe

Just a while ago, I reinstalled the klipper family bucket including Fluidd due to some update compatibility issues. I tried backing up Fluidd configuration files before reinstalling (for example, dashboard layouts, temperature presets, macro categories, etc.), but I couldn't find an option to do this or where to store the configuration files. I had to reconfigure Fluidd after reinstalling, which is actually a rather tedious job, and I may even forget some of the previous settings.

### Describe the solution you'd like

Add a configuration file export option in Fluidd's settings interface to facilitate recovery after reinstallation. Or, friends who know it can tell me what the storage path of the configuration file is, and it is also possible.

pedrolamas, 2023-06-22:
Hi @gaifeng8864, thank you for submitting this Feature Request.

Granted, this is something we can indeed improve on, but in the meantime I suggest you just backup and restore the Moonraker database as a means to easily reinstall!

gaifeng8864, 2023-06-22:
> Hi @gaifeng8864, thank you for submitting this Feature Request.
> 
> Granted, this is something we can indeed improve on, but in the meantime I suggest you just backup and restore the Moonraker database as a means to easily reinstall!

Thank you so much for your recovery. No offense intended, as a user, I don't know much about the underlying operating logic of the klipper family. What data is stored in the moonraker database?

pedrolamas, 2023-06-22:
While Fluidd is a frontend for Klipper, Moonraker sits in the middle as an API server, providing the data and allowing Fluidd to send commands to Klipper.

Moonraker database stores the print history (if you enabled it) and most of the frontend customizations.

You can find some more info here:

https://moonraker.readthedocs.io/en/latest/installation/#data-folder-structure
https://moonraker.readthedocs.io/en/latest/installation/#lmdb-database-backup-and-restore

gaifeng8864, 2023-06-22:
> While Fluidd is a frontend for Klipper, Moonraker sits in the middle as an API server, providing the data and allowing Fluidd to send commands to Klipper.
>
> Moonraker database stores the print [...]

Thank you so much. It's like opening a door to a new world.

pedrolamas, 2023-07-28:
@m-roberts not sure if you are aware of this, but the Creality Sonic Pad does not run Fluidd but an old fork of Fluidd.

Creality broke our license with the Sonic Pad, so you will find no support for it outside of Creality themselves - and the same goes for Klipper, Moonraker, and Klipper screen.

pedrolamas, 2023-08-10:
@m-roberts being very frank here, but I couldn't care less for Creality and the Sonic Pad.

Reason being that if they are pulling Klipper, Moonraker, Fluidd, etc. changes, they are still not updating their forked code, and as such, **they are still in license breach.**

Escrich, 2024-04-15:
Will be a great advantage, to have a plain text file associated to Moonraker database, where, using same Fluidd or mainsail graphic frontend, write the graphical macros definition, and thermal settings too, may be some other items, but for me will be more than enough having those definitions in a text file, and be able to load the definition directly in another differente 3D printer

gaifeng8864, 2024-04-16:
> Will be a great advantage, to have a plain text file associated to Moonraker database, where, using same Fluidd or mainsail graphic frontend, write the graphical macros definition, and thermal [...]

Very good idea. There is a text file that can be directly imported and used in other 3D printers. enough.

PR #1427 (2024-04-30): feat: backup and restore settings
Add a new Backup and Restore buttons to easily export and import Fluidd settings across multiple printers.

As a first implementation, this will backup/restore ALL Fluidd settings (there is no selections of specific parts of the settings).

[image]

Resolve #1113
closed 2024-04-30 by pedrolamas