fluidd-core/fluidd #1131: Fluidd loses homing if filament runs out

AndreasReitberger, 2023-07-31:
### Fluidd Version

v1.24.1

### Browser

Chrome

### Device

Desktop PC

### Operating System

Windows

### What happened

If the filament runs out and the filament sensor triggers, the print cannot be resumed. It first needs homing again.

### What did you expect to happen

The print just resumes

### How to reproduce

Start a print, let the filament run out, and try to resume after the filament has been changed.

pedrolamas, 2023-07-31:
Hi @AndreasReitberger, this seems to me to either be a Klipper (or Klipper config) issue and not something manageable from Fluidd's end, but I would like to confirm.

Can you please upload a klippy.log from your printer from when this issue happened?

AndreasReitberger, 2023-07-31:
Hi @pedrolamas 
Thanks for the quick reply. Sure, the print is active at the moment, the filament should run out the next few hours. Once this has happened, I'll post the log here.

Thanks!

AndreasReitberger, 2023-08-01:
Just made some further tests. When I just pause and extrude the fresh filament using the "Extrude" button on the UI, it works.
Only when I call the "Unload / Load Filament" makro, it looses the homing.
So I rather think it's an issue of the makro, not Fluidd itself.

Sorry for bothering you!

closed 2023-08-01 by AndreasReitberger