fluidd-core/fluidd #1132: Nginx Proxy Manager and Authelia not redirecting as expected

dhenry437, 2023-08-03:
### Fluidd Version

1.24.2

### Browser

Chrome, Firefox, Edge

### Device

Desktop PC, Laptop

### Operating System

Windows

### What happened

I have set up a subdomain to reach my printer through Nginx Proxy Manager and Authelia, I am not always getting the redirect to the Authelia login page.

### What did you expect to happen

For the redirect to take me to Authelia's login page when not authorized.

### How to reproduce

If I go to the subdomain I have configured for fluidd, the first time I get properly redirected to Authelia's login and can authorize and get to fluidd. If I then in another tab go to the auth domain and logout, fluidd expectedly stops working, but refreshing the page does not redirect me to the login as it should, it stays on fluidd and just says cant connect to moonraker. But if you do a hard refresh it works as expected.

### Additional information

I've spent a few days on this and have found the problem. The service worker is catching the page request on a normal refresh and just returning 200, so the server never gets to send its 302. If I go to chrome dev tools it shows that the 200 came from the service worker, also in dev tools going to the worker and ticking `Bypass for network` causes the refresh to redirect to the login page.
[image]
[image]
[image]

dhenry437, 2023-08-03:
I'm not that experience with service workers but these links might be relevant:

https://redfin.engineering/how-to-fix-the-refresh-button-when-using-service-workers-a8e27af6df68
https://redfin.engineering/service-workers-break-the-browsers-refresh-button-by-default-here-s-why-56f9417694

pedrolamas, 2023-08-03:
Hi @dhenry437 thanks for reporting this issue.

We don't directly support such a scenario, but if there is anything we can do to improve/mitigate this issue, we will sure try to do it!

Can you please share your relevant nginx config so I can run some tests on my own machine?

dhenry437, 2023-08-03:
I don't know much about proxy config, but I'm guessing an ideal solution would be if Nginx could somehow tell the browser to bypass the service worker?

Here are the configs from the advanced tab of Nginx Proxy Manager

auth.domain.com

``` nginx
# +-------------------+
# |  Authelia Portal  |
# +-------------------+

location / {
  # START /snippets/proxy.conf;

  ## Headers
  proxy_set_header Host $host;
  proxy_set_header X-Original-URL $scheme://$http_host$request_uri;
  proxy_set_header X-Forwarded-Proto $scheme;
  proxy_set_header X-Forwarded-Host $http_host;
  proxy_set_header X-Forwarded-Uri $request_uri;
  proxy_set_header X-Forwarded-Ssl on;
  proxy_set_header X-Forwarded-For $remote_addr;
  proxy_set_header X-Real-IP $remote_addr;
  proxy_set_header Connection "";

  ## Basic Proxy Configuration
  client_body_buffer_size 128k;
  proxy_next_upstream error timeout invalid_header http_500 http_502 http_503; ## Timeout if the real server is dead.
  proxy_redirect http:// $scheme://;
  proxy_http_version 1.1;
  proxy_cache_bypass $cookie_session;
  proxy_no_cache $cookie_session;
  proxy_buffers 64 256k;

  ## Trusted Proxies Configuration
  ## Please read the following documentation before configuring this:
  ##     https://www.authelia.com/integration/proxies/nginx/#trusted-proxies
  # set_real_ip_from 10.0.0.0/8;
  # set_real_ip_from 172.16.0.0/12;
  # set_real_ip_from 192.168.0.0/16;
  # set_real_ip_from fc00::/7;
  real_ip_header X-Forwarded-For;
  real_ip_recursive on;

  ## Advanced Proxy Configuration
  send_timeout 5m;
  proxy_read_timeout 360;
  proxy_send_timeout 360;
  proxy_connect_timeout 360;

  # END /snippets/proxy.conf;

  set $cors "";

  if ($http_origin ~* (.*[sub1|fluidd|sub2]\.domain\.com)) {
    set $cors "true";
  }

  proxy_pass $forward_scheme://$server:$port;

  if ($cors = "true") {
    add_header 'Access-Control-Allow-Origin' "$http_origin";
    add_header 'Access-Control-Allow-Methods' 'GET, POST, OPTIONS, DELETE, PUT';
    add_header 'Access-Control-Allow-Credentials' 'true';
    add_header 'Access-Control-Allow-Headers' 'User-Agent,Keep-Alive,Content-Type';
  }
}
```

fluidd.domain.com
[... 96 lines cut ...]
  ## Uncomment this line if you're using NGINX without the http_set_misc module.
  # set $target_url $scheme://$http_host$request_uri;

  ## Save the upstream response headers from Authelia to variables.
  auth_request_set $user $upstream_http_remote_user;
  auth_request_set $groups $upstream_http_remote_groups;
  auth_request_set $name $upstream_http_remote_name;
  auth_request_set $email $upstream_http_remote_email;

  ## Inject the response headers from the variables into the request made to the backend.
  proxy_set_header Remote-User $user;
  proxy_set_header Remote-Groups $groups;
  proxy_set_header Remote-Name $name;
  proxy_set_header Remote-Email $email;

  ## If the subreqest returns 200 pass to the backend, if the subrequest returns 401 redirect to the portal.
  error_page 401 =302 https://auth.domain.com/?rd=$target_url;

  # ! END /snippets/authelia-authrequest.conf;

  proxy_pass $forward_scheme://$server:$port;
}
```

All the subdomains are working ie. go to the right place, all switches are turned on (`Cache Assets`, `Block Common Exploits`, `Websockets Support`, `Force SSL`, `HTTP/2 Support`, `HSTS Enabled`, `HSTS Subdomains`), all subdomains have a valid SSL Certificate. As I mentioned Authelia is working as intended on other protected domains.

dhenry437, 2023-08-03:
I may have inadvertently found a hacky solution, I just blocked the browser from fetching `/sw.js`. It does work as intended now. Although I realise this completely blocks offline use of fluidd

pedrolamas, 2023-08-03:
Well, given you want to add auth, I guess offline access won't be possible anyway!

dhenry437, 2023-08-03:
my thoughts exactly, I can just use the local address for my pc here when the net cuts out.

but this solution came about when I added

``` nginx
location = /sw.js {
    add_header Cache-Control 'no-store, no-cache';
    if_modified_since off;
    expires off;
    etag off;
}
```

to try and see if I could force the service worker to avoid the cache? what are your thoughts on this?

pedrolamas, 2023-08-05:
7130 is Moonraker, correct? If so, your best choice there is adding it to nginx under the same url as Fluidd (though that means single instance!)

Here's my own Nginx config with Fluidd using that approach:

https://github.com/pedrolamas/docker-iot-stack/blob/master/nginx/etc-nginx-templates/nginx.conf.template#L269C22-L276

pedrolamas, 2023-09-20:
Closing this now as there has been no more contact regarding this matter.

If you still need help with this, I would recommend that you try [Klipper Discord server](https://discord.klipper3d.org/) instead.

closed 2023-09-20 by pedrolamas