fluidd-core/fluidd #1136: Simplify translation process with Weblate

Daviteusz, 2023-08-06:
### Is your feature request related to a problem? Please describe

Translators are not developers and may have trouble using the current translation method (vscode with addon or babeledit) They may even have problems with github usage.

### Describe the solution you'd like

I use self-hosted Weblate for translation. It has a very convenient gui, you can set various flags, for example, mark untranslatable words or characters that you can click on and they will be automatically filled in the translation field.

Changes can be sent automatically e.g. every 24h as PR or as commit. Commit descriptions or PR names can be managed by the project author, not the translator. The translator can simply focus on the translation.

I've been using this solution for a few months now and overall I haven't experienced any problems. 

### Additional information

[image][image][image]<img scr=https://github.com/fluidd-core/fluidd/assets/11527277/cb5ee132-e2ae-4208-b144-a897ab7f71d5 width=50%>[image]

Daviteusz, 2023-09-11:
Pull request #1164 has appeared, so...

pedrolamas, 2023-09-11:
Thanks @Daviteusz but it's not a done deal yet... we are still trying to get Fluidd licensed as a Libre project!

Daviteusz, 2023-11-17:
Closing due to #1239

closed 2023-11-17 by Daviteusz