fluidd-core/fluidd #1137: Incorporation of Moonraker [sensors]

JamesH1978, 2023-08-07:
### Is your feature request related to a problem? Please describe

Moonraker has the ability to register sensors from other devices via MQTT and other methods,

### Describe the solution you'd like

 It would be nice to see these in the Fluidd interface

pedrolamas, 2023-09-07:
I've taken a quick look at this, I guess the main question from me is how we are to present dynamic data...

Users can use `set_result` to pass not only temperature, pressure, humidity (all measurements that Fluidd can already show on the temperature chart), but also any other key they could want (like power, energy, etc.)

We might need to use some table/tree view to show the values from these extra sensors!

bartwaw, 2023-09-27:
but can't be done like this: total watts used by print job, and then store it in some log. I'm not best in programming

JamesH1978, 2023-09-27:
Fkuidd has no log, everything it can store at the moment is held in the moonraker database, which would be up to that project to implement first

bartwaw, 2023-09-27:
So as i understand for example  "Print duration" is stored in moonraker ?

closed 2023-10-03 by pedrolamas

midnight1979, 2024-01-30:
> @bartwaw adding to history will probably not be possible, as that would require Moonraker to store history on that side first!
> 
> If we did that on Fluidd only, once you close the tab or refresh it, all the history data would be cleared!

As I know now moonracker having this feature and store sensor values history. Does it possible to add these values to chart?

PR #1196 (2023-10-03): feat: adds Moonraker sensor support
Initial support for Moonraker `[sensor]`.

[image]

Note: Moonraker provides sensor history via `server.sensors.measurements` endpoint and we might eventually add it in the future, but for now this PR only shows the current state.

Resolves #1137