fluidd-core/fluidd #1142: Camera in dashboard shows broken image icon, stream works fine.

setral, 2023-08-10:
### Fluidd Version

1.25.0

### Browser

Firefox

### Device

Desktop PC

### Operating System

Windows

### What happened

Updated Fluidd on two different Pi setups.

After the UI reset, the Cameras widget on the dashboard was showing a broken image/link icon instead of the camera view.
I could go to the stream and see the Camera without issue.

After chatting in the Voron Discord, went into the Camera settings and noticed there was a new "Camera Url Snapshot" field that was blank. After setting it to /webcam?action=snapshot and clicking save, the Camera widget on the dashboard was working again.

Pi Zero 2 W with a USB Camera on a modified Ender 3 Pro
And
Pi3A+ with a Pi Camera on a Voron v0.1

### What did you expect to happen

I expected the Cameras widget to still function after the update.

### How to reproduce

Updated Fluidd from Settings \ Software Update.
Fluidd restarted.
Camera widget no longer displaying.

Fixed by going into Settings \ Cameras and modified my camera, noticed the new blank "Camera Url Snapshot" field, and modified it to: /webcam?action=snapshot

### Additional information

[image]
[image]

pedrolamas, 2023-08-10:
Hi @setral, thank you for reporting this, and I can confirm you are absolutely correct.

Before Fluidd v1.25.0, we only asked for a single address for the camera and would then infer from that what the stream URL and snapshot URL would be (we would suffix the URL with `?action=stream` or `?action=snapshot` accordingly)

However, we have recently found out that this was causing issues with Crowsnest, and as such dropped that approach to a more manual indication of the camera addresses, which is what you see now.

As we have no automatic migration process in place for this change, I've now added some notes in the [Discord Release Announcement](https://discord.com/channels/762863525818662974/793545946755366933/1137882619992879286) and on the [GitHub Release](https://github.com/fluidd-core/fluidd/releases/tag/v1.25.0) to reflect this change and the manual steps to fix it.

closed 2023-08-10 by pedrolamas

PKav89, 2023-08-11:
Hello!
In my case it did not help. But "broken image" url links to http://192.168.1.22:7125/webcam?action=snapshot, but correct port is 80 and there are indeed stream and snapshot. How to remove 7125 port from image url?

pedrolamas, 2023-08-11:
@PKav89 just use the full URL (with http, host, port, path, etc), the whole thing!

PKav89, 2023-08-11:
Huh, it worked! Thanks!
But I'm curious where did 7125 came from? There's no such port in crowsnest or fluidd config.

pedrolamas, 2023-08-11:
That's the default Moonraker port! 🙂

Having said that, still a bit of a but there, as it should not use the moonraker port but the browser one (I'll fix that later)

Creo2005, 2023-08-11:
Camera

adamo000, 2023-08-12:
I have still a problem with the camera preview on dashboard on fluidd v1.25.1 with no updates available. 

**The camera preview looks like this**

[image]

**With settings:**
[image]

**I've try another method to see if the raspberry pi camera v2 is working and it does.**
```
pi@raspberrypi:~ $ libcamera-jpeg -o test.jpg
Preview window unavailable
[0:06:34.801098542] [3705]  INFO Camera camera_manager.cpp:297 libcamera v0.0.5+83-bde9b04f
[0:06:34.871229912] [3706]  WARN RPI vc4.cpp:383 Mismatch between Unicam and CamHelper for embedded data usage!
[0:06:34.872997567] [3706]  INFO RPI vc4.cpp:437 Registered camera /base/soc/i2c0mux/i2c@1/imx219@10 to Unicam device /dev/media4 and ISP device /dev/media0
[0:06:34.873095622] [3706]  INFO RPI pipeline_base.cpp:1101 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
[0:06:34.875029832] [3705]  INFO Camera camera.cpp:1033 configuring streams: (0) 1640x1232-YUV420
[0:06:34.875985807] [3706]  INFO RPI vc4.cpp:565 Sensor: /base/soc/i2c0mux/i2c@1/imx219@10 - Selected sensor format: 1640x1232-SBGGR10_1X10 - Selected unicam format: 1640x1232-pBAA
#0 (0.00 fps) exp 10926.00 ag 2.00 dg 1.05
#1 (30.01 fps) exp 10926.00 ag 2.00 dg 1.04
#2 (30.01 fps) exp 11304.00 ag 2.00 dg 1.01
#3 (30.01 fps) exp 11436.00 ag 2.00 dg 1.00
#4 (30.00 fps) exp 11398.00 ag 2.00 dg 1.00
.
.
.
#140 (30.00 fps) exp 11379.00 ag 2.00 dg 1.00
#141 (30.00 fps) exp 11379.00 ag 2.00 dg 1.00
#142 (30.02 fps) exp 11379.00 ag 2.00 dg 1.00
[0:06:40.004321311] [3705]  INFO Camera camera.cpp:1033 configuring streams: (0) 3280x2464-YUV420 (1) 3280x2464-SBGGR10_CSI2P
[0:06:40.006252187] [3706]  INFO RPI vc4.cpp:565 Sensor: /base/soc/i2c0mux/i2c@1/imx219@10 - Selected sensor format: 3280x2464-SBGGR10_1X10 - Selected unicam format: 3280x2464-pBAA
Still capture image received
pi@raspberrypi:~ $
```
[image]

Here is my /boot/config.txt
```
# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
dtparam=spi=on

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
#dtparam=audio=on

# Automatically load overlays for detected cameras
#start_x=1

# Automatically load overlays for detected DSI displays
#display_auto_detect=1

# Enable DRM VC4 V3D driver
#dtoverlay=vc4-kms-v3d
#max_framebuffers=2

# Run in 64-bit mode
arm_64bit=1

# Disable compensation for displays with overscan
disable_overscan=1

[cm4]
# Enable host mode on the 2711 built-in XHCI USB controller.
# This line should be removed if the legacy DWC2 controller is required
# (e.g. for USB device mode) or if USB support is not required.
otg_mode=1

[all]

[pi4]
#dtoverlay=vc4-fkms-v3d
# Run as fast as firmware / board allows
arm_boost=1
# scale voltage up on demand for over clocking  

[all]
gpu_mem=256
hdmi_enable_4kp60=1
dtoverlay=imx219
```

What can I do to make the preview be available?

pedrolamas, 2023-08-12:
@adamo000 you are getting "no signal" which tells me the stream connection in Fluidd is working as that message is from the stream (not a Fluidd message).

I recommend that you check your Crowsnest log for issues, and if that doesn't help, jump to Klipper discord server, and ask for help on the `#webcams` channel.

adamo000, 2023-08-12:
I uninstalled crowsnet and the problem exist. 
The config hasn't changed, a broken icon is present on Dashboard and after try to enforce full screen, I've got 502 Bad Gatway error.
[image]

It looks that there is still something bad here. Is there anything that I can do to help you to solve the issue?

pedrolamas, 2023-08-12:
I have no idea what streamer you are using for your webcam... The recommended way to get a webcam working in Fluidd or Mainsail is via Crowsnest.

GitHub is not the place for this type of query, my advice is that you try Klipper discord server for further assistance.

Shadow-FW, 2023-08-12:
Update 1.25 broke my cams too. Add snapshot section not help. Clean URL of stream have image.

pedrolamas, 2023-08-12:
@Shadow-FW try updating to 1.25.1, if the problem persists, please post an image or your camera config