fluidd-core/fluidd #1147: Add Used Filament Stats to Main Screen

Shadow-FW, 2023-08-12:
### Is your feature request related to a problem? Please describe

Is not a problem

### Describe the solution you'd like

Add Used Filament Stats to Main Screen, around Printing or Tool part. And add button to reset only total_filament_used, without reset other print stats.

pedrolamas, 2023-08-15:
The job history totals are maintained by Moonraker, and currently there is no option to selective reset, only to reset the whole thing...

Shadow-FW, 2023-08-15:
> The job history totals are maintained by Moonraker, and currently there is no option to selective reset, only to reset the whole thing...

Ok. But can still display "Total_filament_used" on the main screen?

pedrolamas, 2023-08-15:
We will probably be looking into some changes on the main dashboard screen, so we might check what we can do about this then!

closed 2025-04-06 by pedrolamas